prusa3d/Original-Prusa-i3 #1: Spool holder part problems

poyaque-estamos, 2016-04-21:
Hi!

The stl for the spool holder doesn't match the one that comes with the printer. It doesn't have holes in it to save plastic when printing out. Also, there is not two versions for each side of it, just one. These problems are also seemly present in the scad file, considering the dxf doesn't have the holes and there are no comments in the scad file to get right-left version.

poyaque-estamos, 2016-04-21:
I have fixed the problem with left and right spool holder myself, in the scad file, but the holes would be nice to get from your model. I'll add it as a pull request.

poyaque-estamos, 2016-06-14:
Seems the left and right arms for the spool holder are included in the mk2 branch. Good! Still no holes on the dxf and therefore the stl files.

prusa3d-bb, 2016-06-20:
Hello,
yes, I added it to MK2 branch. Supplied spool holders are now shipped without holes so it’s consistent with our manufacturing process.
If you want the spool holder with holes, please use attached dxf and replace it with MK3 version.

—
Michal Průša, CTO

closed 2017-12-05 by akukan