prusa3d/Original-Prusa-i3 #6: Feature request: Offer setting to change the brightness of the LCD screen

mrjohnc, 2016-09-26:
It would be nice to have this option, especially for printing at night when the screen can be very bright

ManuGithubSteam, 2016-09-27:
I second this! I also asked support about it. It would be good to turn it off into "night" mode so only a knob turn turns it back on for a few seconds.

bubnikv, 2016-09-27:
Yeah! That would be great! Only the current hardware does not support this.

mrjohnc, 2016-10-02:
Ok, thanks @bubnikv, shame its not possible. Is it the particular screens that are shipped or the controller board?

ManuGithubSteam, 2016-10-02:
Would it be possible to turn it off completely ?

mrjohnc, 2016-10-02:
Ah yes, this would be good, maybe an option in settings for 'display sleep after 5 minutes'? and any movement in the dial would turn the brightness back on again.

bubnikv, 2016-10-02:
If you know which side of the soldering iron is hot, you can surely
disconnect the display or add a manual switch to the backlight. But to
control it by the software is another thing, you would need to add a data
line from the RAMBo to the displa, add a transistor and likely a resistor
or two. It is doable, but not that easy.

mrjohnc, 2016-10-03:
@bubnikv so this could not be done by software alone?

bubnikv, 2016-10-03:
That is correct. It cannot be controlled by the software alone. You have to do some soldering job on the display board.

mrjohnc, 2016-10-04:
Ok, hopefully a display capable of this will be added to future versions of the kit

boelle, 2017-12-04:
i think its safe to close this one as the hardware is not able to do it.

and it has been so for a year and its very unlikely to change anytime soon

closed 2017-12-05 by akukan