prusa3d/Original-Prusa-i3 #7: Silent mode isn't very quiet when the print head is travelling

mrjohnc, 2016-09-27:
When extruding the printer is quiet but when the print head is traveling the printer is loud, I have tried turning the travel speed down but it doesn't seem to work, maybe acceleration? 

Is it possible to limit maximum acceleration and print head travel speed during silent mode?

prusa3d-bb, 2016-10-01:
Hi,

You can also check your Y axis rod alignment. The noise can be caused by your bed vibrating due to slight misalignment of the Y axis.
Also the table on which the printer is set can be a problem. It it wobbles, there is also great possibility of vibrations caused by the table itself.
After checking these points you should have pretty good noise levels. But note that the printer will never be completely silent while printing.

Take care

—
Dan Oravsky
Customer Support

PRUSA Research
+420 222 263 718 
621/8a Prvniho pluku, 18600, Prague
shop.prusa3D.com

--- original message ---

--- end of original message ---

mrjohnc, 2016-10-02:
Ok, thanks

closed 2016-10-02 by mrjohnc