prusa3d/Original-Prusa-i3 #8: Add option to beep when finished print

mrjohnc, 2016-10-02:
Add option in settings menu for the printer to beep when a print has finished. This will help to get the attention of the user if they are not in the room, listening to music etc.

prusa3d-bb, 2016-10-07:
Hello John,
Thanks for your feedback!

Actually, we have this for long time in our list and still thinking how to do it to satisfy all people.

We'll definitelly do that!

—
Michal Průša, CTO

--- original message ---

> Add option in settings menu for the printer to beep when a print has finished. This will help to get the attention of the user if they are not in the room, listening to music etc.

--- end of original message ---

AJMartel, 2017-09-14:
I was looking to add this option as well.
For now a workable solution is to add the following to your "End gcode":

M300 S1000 P50		       ;First beep
M300 S0 P250		       ;Wait
M300 S1000 P50		       ;Second beep

stelgenhof, 2018-05-27:
Since this repository concerns the 3D printed parts, shouldn't this issue be transferred to the firmware or Slic3rPE repository?

richiewebgate, 2019-08-11:
one year later:
any update on "beep when done"? Just received a new MK3S and I can't find an option to alert me when printing is completed :)

crisoberillo, 2020-01-03:
> 
> 
> Hello John,
> Thanks for your feedback!
> 
> Actually, we have this for long time in our list and still thinking how to do it to satisfy all people.
> 
> We'll definitelly do that!
> 
> Michal Průša, CTO
> 
> --- original message ---
> 
> > Add option in settings menu for the printer to beep when a print has finished. This will help to get the attention of the user if they are not in the room, listening to music etc.
> > --- end of original message ---

Hello Michal,

I'm looking forward to see this option implemented.
Do you have any news about it?

Thank you in advance,
Andrea

DB3DE, 2020-01-04:
> > Hello John,
> > Thanks for your feedback!
> > Actually, we have this for long time in our list and still thinking how to do it to satisfy all people.
> > We'll definitelly do that!
> > Michal Průša, CTO
> > --- original message ---
> > > Add option in settings menu for the printer to beep when a print has finished. This will help to get the attention of the user if they are not in the room, listening to music etc.
> > > --- end of original message ---
> 
> Hello Michal,
> 
> I'm looking forward to see this option implemented.
> Do you have any news about it?
> 
> Thank you in advance,
> Andrea

Dear all,
there is a super easy way to do it already there.
Edit the Slicer setting and let it add the appropiate G-Code setting at the end of all generated G-Code files. I would recommend the following snipped:

;this snipped plays a kind of imperial march
;great for the end of your print job :-)
;imperial martch
M300 S220 P300 ;LA3
G4 P300
M300 S220 P300 ;LA3
G4 P300
M300 S220 P300 ; LA3
G4 P300
M300 S174.61 P350 ;F3
G4 P75
M300 S261.63 P250  ;c4
G4 P100
M300 S220 P350
G4 P100
M300 S174.61 P350 ;F3
G4 P75
M300 S261.63 P250  ;c4
G4 P100
M300 S220 P400

;-)

stelgenhof, 2020-01-04:
To repeat: this repository concerns the printed parts and not the firmware. This issue should be transferred to the firmware repository. That way chances of this issue being picked are much higher.

Cheers! Sacha

mrjohnc, 2020-01-05:
> To repeat: this repository concerns the printed parts and not the firmware. This issue should be transferred to the firmware repository. That way chances of this issue being picked are much higher.
> 
> Cheers! Sacha

Hi Stelenhof, how do I do this? Do I just need to make a new one or can I move it somehow?

stelgenhof, 2020-01-05:
@mrjohnc You just need to create a new issue at the firmware repository as there is no possibility to automatically transfer an issue on github.com. (Apologies if I made you believe that) :)

Cheers! Sacha

akukan, 2020-01-05:
Hi @mrjohnc ,
as this is firmware-related issue, let me transfer it to the prusa3d/Prusa-Firmware repository.