prusa3d/Original-Prusa-i3 #9: Unable to preheat printer below 15 degrees C

StefanL2010, 2016-11-06:
PROBLEM:
If printer is started at an ambient temperature below 15 degrees Celsius it immediately triggers MINTEMP error on both the bed and hotend, even before any attempts to heat the bed or hotend.
This is because MINTEMP is set at 15 degrees in the firmware settings(usually 5 degrees on other printers). MINTEMP is a thermal runaway protection since a faulty thermistor usually sends a low value and should trigger a shutdown of the heater to prevent thermal runaway.
This also means that the printer can not be used below 15 degrees.

SUGGESTION:
No need to trigger MINTEMP error if the heater isn't enabled, thermal runaway isn't an issue when the heater is disabled.
When starting Preheat of the printer disable MINTEMP error for x amount of time to enable the bed and hotend to reach temperatures over 15 degrees.
Alternately add an option somewhere to trigger power to the bed and hotend for 10 seconds manually(overriding the MINTEMP error). That way you can manually add bursts of power to get the heat up.

Grogyan, 2016-11-06:
Or Just lower the threshold to something lower like 5 degrees C

StefanL2010, 2016-11-06:
Well, the question is if Prusa set it to 15 degrees for a reason or not. Perhaps they had some of their thermistors go bad and showing more than 5 degrees. If I just lower the MINTEMP without knowing this I could potentially risk a thermal runaway issue. The best way would be to still keep the 15 degree MINTEMP but still be able to preheat the printer.

Grogyan, 2016-11-06:
Are you sure you don't have a broken thermistor?

StefanL2010, 2016-11-06:
Yes, both the hotend and bed works perfectly. The MINTEMP is set at 15 degrees in Prusas firmware so it's only logical that the printer triggers a MINTEMP error below 15 degrees (since the printer is in a garage with an ambient temperature below 15 degrees). If anything it means the the termistors work as intended and are reading a correct temperature.
I'm looking for a workaround to be able to preheat the printer below 15 degrees even though the MINTEMP is set at 15 degrees.

prusa3d-bb, 2016-11-08:
I understand now :) We do suggest keeping the printer in the warm so that it avoids enviro factors like drafts and cold. I can link you to our firmware source files if you'd like to adjust them to a new setting.. I'd be very cautious about doing this. The other alternative would be to build an enclosure for your printer.

_________

Greyson Haswell
Prusa Research
Prague, Europe
info@prusa3d.com

--- original message ---

--- end of original message ---

StefanL2010, 2016-11-08:
I can modify the firmware and change the MINTEM setting myself. But I'd rather not do that because it might be set to 15 degrees for a reason. Lowering the setting might cause thermal runaway issue. I already have the printer in an enclosure and the bed of the printer quickly heats up the enclosure. The problem is getting it going and starting the heater, once the heater is started the ambient temperature rises in the enclosure.
The FAQ at prusa3d.com states that the printer waits 45 seconds before triggering a MINTEMP error when the printer is operating (if a thermistor shows less than 15 degrees). The part that I consider a bug is that MITEMP error triggers directly when i power the printer on. There is no reason to trigger the MINTEMP error when the heaters are disabled.

bubnikv, 2016-11-08:
The thermistor reading range is set by a resistor divider at the RAMBO
board. At 15 degrees the voltage at the termistor is at its limit.
Therefore the RAMBO board cannot effectively measure lower temperatures
than what you read on the display.

The recommended workaround is to preheat the printer with a fan or a heat
gun. But be careful with a heat gun :-)

StefanL2010, 2016-11-08:
Well thats not true since the ambient temp around my printer was 5 degrees at startup last night and both the hotend and the bed both correctly read 5 degrees. So the thermistors have no problems reading lower values than 15 degrees. But thats missing the point.

1. The MINTEMP error is a thermal runaway protection only, Yes?
2. If Yes, why does the error trigger even though the heaters are not enabled?
3. The FAQ states there should be a 45 sec delay before MINTEMP triggers(if the temp is lower than 15 degrees). This is not true since it triggers directly at printer startup(even though heaters are not enabled).

prusa3d-bb, 2016-11-11:
Hello Stefan, 

That's not entirely accurate. MINTEMP and Thermal runway can be related, but not always. If it's triggering right away, it can simply be a generic failure of the thermistor. In which case; send us a video of the instant flash up of the failure and we should be able to send you a replacement sent out to you on Monday after verification.

_________

Greyson Haswell
Prusa Research
Prague, Europe
info@prusa3d.com

--- original message ---

The MINTEMP error is a thermal runaway protection only, Yes?If Yes, why does the error trigger even though the heaters are not enabled?The FAQ states there should be a 45 sec delay before MINTEMP triggers(if the temp is lower than 15 degrees). This is not true since it triggers directly at printer startup(even though heaters are not enabled).

bardiir, 2017-07-17:
You can just grab the hotend nozzle for a few seconds until it gets over 15°C and then heat up normally from there. I wouldn't really change anything in the software, the limit is there for a reason anyway.

closed 2017-12-05 by akukan