prusa3d/Original-Prusa-i3 #10: Move socket cutout on back of PSU cover

pboschi, 2016-11-20:
The socket cutout on the PSU cover should be put on the rear, not on the right side. There's plenty of unused space on the back of the printer when it sits on a desk (due to the heatbed cables), so there's no reason to have the power plug waste additional space on the right.

prusa3d-bb, 2016-11-20:
Hello

The placement of the power socket was logical based on the soldering work required for the PSUs, a simple change like moving the placement back increases the complexity of the soldering work. It will also increase the production time for all PSUs, and at this time we can't afford to do that.

_________

Greyson Haswell
Prusa Research
Prague, Europe
info@prusa3d.com

--- original message ---

--- end of original message ---

pboschi, 2016-12-09:
BTW, here is a little mod to put the power socket on the rear:
http://www.thingiverse.com/make:274708

closed 2017-12-05 by akukan