prusa3d/Original-Prusa-i3 #11: Y motor holder a bit flimsy

pboschi, 2016-11-20:
The Y motor holder is a bit flimsy and bends, reducing tension and misaligning the belt. It should be integrated with the Y-distance part from the start, and come with 3 screw points instead of 2.

MartinMajewski, 2016-11-21:
Please take a look at my holder design on Thingiverse: http://www.thingiverse.com/thing:1837853

pboschi, 2016-11-21:
Thanks, i'll try that. The fact is that, depending on material and print quality, the additional bracket alone may or may not reduce the bending of the motor when the belt is finally installed and tightened. If not, you would still need the "Y-distance" part from this repository. With 3 brackets you could now install a damper, but the motor would be coupled to the rest of the printer by that "Y-distance" part, making the damper less effective.

I think that your design, slightly modified to be a bit wider (at least the lower "bracket"), would definitely solve the problem, and give the option to install a damper (with full effectiveness).

Just look at the X motor bracket... it doesn't bend at all because it has 3 screws AND it's wider.

MartinMajewski, 2016-11-21:
Please tell me what "Y-distance" part you mean exactly. I have searched the repository but I am not sure what you mean. Is it the 2-point Y-motor holder?

From my experience the 3-point motor holder is absolutely sufficient to hold the motor in position. I have printed it with PLA, three perimeters and 20% honeycomb infill and this is already overkill. But you can always print it sturdier.
The belt should never be that tight, that is bends the motor, anyway. It is absolutely enough if it does not slip over the pulley's teeth when you give it a firm pull. As soon as the motor starts bending (with a dampener or  without), you have tighten the belt too much. This puts unnecessary stress on the motor's shaft.

pboschi, 2016-11-21:
I'm referring to this part:
https://github.com/prusa3d/Original-Prusa-i3/blob/MK2/Printed-Parts/scad/Y-distance.scad

The belt is not over-tightened; but after some days of use i found a bending of the Y motor mount, which in turn causes the belt to slip on a side; that's a very minor problem i know, but with just some more width it would be perfect (but i'll just try with higher infill for now, as you suggested).
[image]

prusa3d-bb, 2016-11-24:
Hello, 

It could possibly be resolved by extending the pulley on the shaft a little, so that it better aligns with the y-holder. It might help, while you're reprinting the part.

_________

Greyson Haswell
Prusa Research
Prague, Europe
info@prusa3d.com

--- original message ---

https://github.com/prusa3d/Original-Prusa-i3/blob/MK2/Printed-Parts/scad/Y-distance.scad

--- end of original message ---

pboschi, 2016-11-26:
These two things may help to fix the problem:
http://www.thingiverse.com/thing:1911701
http://www.thingiverse.com/thing:1914857

closed 2022-05-05 by Prusa-Support