prusa3d/Original-Prusa-i3 #12: Printer allows unloading of filament below PLA melting temperature with no warning

mrjohnc, 2016-11-21:
I managed to clog my printer by accidentally pressing unload filament before the printer had reached the preheat temperature for the material I was using (PLA). This caused filament to get stuck in the nozzle further up and I had to take the print head apart to fix it.

Maybe add in a control to say only allow unload filament unload once printer has reached preheat temperature 

I'm using 3.0.8

prusa3d-bb, 2016-11-22:
Hi John,

You can clean the hotend following those steps:
preheat the extruder to 260C and extrude some of the filament.
if extruding won't help, use the acupuncture needle and stick it up the nozzle from bottom.
the last option is to get a 1.5mm steel wire and insert it into the extruder instead of filament and then push it down.

Best regards

—
Dan Oravsky
Technical Support

PRUSA Research
+420 222 263 718 
621/8a Prvniho pluku, 18600, Prague
shop.prusa3D.com

--- original message ---

--- end of original message ---

mrjohnc, 2016-11-22:
Hi Dan

I'm happy to say that I fixed the issue with help from Prusa support. My bug is related to the software issue that allowed this to happen. Its very easy to do by accident and takes quite a long time to fix the first time it happens.

Thanks

John

prusa3d-bb, 2016-11-23:
Hello John,

I just want to check that all is good now and you are able to print.

Let me know and have a nice day!

Monika Dostalova
Customer Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
621/8a Prvniho pluku, 18600, Prague
shop.prusa3D.com

--- original message ---

mrjohnc, 2016-11-23:
Yes, absolutely fixed, I wasn't filing this about getting my printer fixed, just pointing out a software issue

Thanks

bardiir, 2017-07-17:
Actually I would like to see this as an advanced option to explicitly enable this to enable atomic pulls supported by the extruder motor instead of just yanking at the filament. Atomics are still the best way to get dirt out of the nozzle, although I'm pretty happy to say that I've managed to get over 100h print time already and not clog the nozzle yet thanks to the filament filter any taking care not to drop dirt into the feeder :)

mrjohnc, 2017-07-17:
> Actually I would like to see this as an advanced option to explicitly enable this to enable atomic pulls supported by the extruder motor instead of just yanking at the filament. Atomics are still [...]

Yes this makes sense, to have it blocked when trying to unload filament normally when it is under temp and then a special option for atomic pull which would have an option for each filament type. The printer would then guide the user through the atomic pull process with clear instructions heating and cooling the hot end as needed.

closed 2022-05-05 by Prusa-Support