prusa3d/Original-Prusa-i3 #17: Prevent oozing from the nozzle before calibration point check messing up your calibration

ManuGithubSteam, 2016-11-24:
Hello Prusa,

I noticed the following behavior:

When the nozzle heats up, the material is oozing out of the nozzle. This is normal due to heating. Note: I'm using the prusa profiles and material so everything should be ok in the temp. regard.

So before the 9 calibration recheck, i have to manually remove the oozed out filament to not mess up the calibration point recheck.

It would be easy i assume to put in a little gcode to wipe the nozzle a bit before the calibration recheck starts. Also a little retraction milli seconds before the calibration touches the bed will help. This way the ozzing while the check is minimised and the print is not messed up.

What do you think ?

Thanks!

pboschi, 2016-11-26:
i would change the default order of actions:
1) heat bed
2) 9-point calibration
3) go to 0,0,20 (so it's easy to clean manually)
4) heat nozzle
6) start

bubnikv, 2016-11-27:
>    5. start

The nozzle is hot at the time of the 9 point calibration intentionally. If
it was not hot, any leaked plastic at the tip of the nozzle would push the
bed from the probe making the calibration unreliable.

bardiir, 2017-07-17:
I'd suggest retracting a cm or so instead, this pulls out the filament from the hot zone and should prevent most of the oozing while measuring.

bubnikv, 2017-07-18:
> I'd suggest retracting a cm or so instead, this pulls out the filament
from the hot zone and should prevent most of the oozing while measuring.

At the start of the print, we don't know the position of the filament in
the melt zone. If you retract the filament out of the heat break, you may
clog the extruder by letting the molten filament expand where it should not
expand. This is exactly the issue all the single-extruder multi-material
setups face and we had to develop quite an extensive hardware / software
system to minimize the clogs to an acceptable level.

bardiir, 2017-07-18:
I'm currently not aroung the printer - can it physically move outside the bed with the nozzle? If so maybe do it like this:
Move outside bed perimenter
Extrude 5mm
Retract 5mm
Wait a few seconds
Wipe at the edge of the bed
Do calibration moves
Reinsert Filament 5mm
Prime & Print

If it can't move outside the bed perimeter maybe prime the nozzle on the right side of the bed first and then retract.
This way you don't have any risk of pulling out the filament too far as you insert it a controlled amount first, so if it wasn't in the wrong position to begin with it will not be in a wrong position afterwards.

fucx, 2017-07-18:
You could just do a retract before mesh bed levelling in your start gcode like the S3D config files by philipp.m on the forums: http://shop.prusa3d.com/forum/original-prusa-i3-mk2-f23/simplify3d-print-settings-august-30-2016--t938-s200.html#p28288

He's running this start gcode which keeps the nozzle from oozing:

```
G28 W ; home all axes without mesh bed leveling
G1 F1000.0 ; set feed rate
G1 E-10 ; retract to avoid blobs on the bed
G80 ; run mesh bed leveling routine
G1 Y-3.0 ; prepare to prime
G1 E15 ; bring back the filament
G1 X60.0 E9.0 ; priming
G1 X100.0 E12.5 ; priming
```

bardiir, 2017-07-18:
@fucx This was my first idea too but as @bubnikv has explained, this could potentially clog the extruder as it uncontrolled pulls out the filament from an unknown position, so you need to establish a known position first or at least make sure you don't end up in a worse position than you started.

fucx, 2017-07-18:
@bardiir you're right. but keeping that in mind, it shouldn't be too much of a problem. it's not a general solution, of course. but if you're aware of the pitfalls, it should work.

bardiir, 2017-07-18:
@fucx yes but if there are any pitfalls it should not be included in the general Firmware of the printer to not cause any issues with inexperienced users. Advanced users can at any time manually add stuff like this to the slicer anyway. Blobs on the printbed don't cause prints to fail generally and the user should have the most stable solution possible at hand by default.

boelle, 2017-12-04:
i think this is more a cosmetic issue that a functional isuue... a few drops of filament should not matter

and as @bubnikv says it has to be hot or else filament that sits on the tip of nozzle will make the probing invalid

i think this one should be closed

closed 2017-12-05 by akukan