prusa3d/Original-Prusa-i3 #18: Extrusion Immediately After Color Change

lbussy, 2017-01-16:
When a color change is executed and you select "yes" to continue, the nozzle will immediately extrude ~40mm filament into mid-air and then move to the print.   I believe the offending section of GCode is this:

`M600`
`G1 E0.80000 F2100.00000`

I see no reason for the extrusion after the color change, and if one does not catch it and pull it off before printing resumes, the filament strand will embed in the print.

lbussy, 2017-01-16:
@cahe on Slack looked at this and found that the extrusion in GCode is there before and after.  He did find this portion of the firmware that seems likely on line 4494 of Marlin_main.cpp:

`//Not let's go back to print`

`     //Feed a little of filament to stabilize pressure`
`     target[E_AXIS]+= FILAMENTCHANGE_RECFEED;`
`     plan_buffer_line(target[X_AXIS], target[Y_AXIS], target[Z_AXIS], target[E_AXIS], FILAMENTCHANGE_EXFEED, active_extruder);`
`       `
`     //Retract`
`     target[E_AXIS]+= FILAMENTCHANGE_FIRSTRETRACT;`
`     plan_buffer_line(target[X_AXIS], target[Y_AXIS], target[Z_AXIS], target[E_AXIS], FILAMENTCHANGE_RFEED, active_extruder);`

This seems a likely culprit.  I'm not sure if this extra extrusion and then retraction is needed, or if something is causing it to extrude more than intended.

bubnikv, 2017-01-16:
The extra extrusion is intended. If you wait for a while to confirm that the filament has been exchanged successfully, some of the material may leak out of the nozzle, leading to a gap in a print. The extra extrusion has the purpose to pressurize the nozzle before resuming the print.

lbussy, 2017-01-16:
I can appreciate that a fresh extrusion may be desirable to pressurize the nozzle.  The practical issue however is that there is a piece of filament hanging off the nozzle as it goes to resume printing.  Without something like a "purge tower" to clean it, a user is left having to time cleaning the nozzle as it moves to print.  

Have you personally tried color printing?  The experience will do more to understand the implications than I can hope to type here.   A low-height print makes it pretty difficult; it's possible someone might jam a tool (or worse a finger) against a hot print head while trying to clean that filament off.

lbussy, 2017-01-16:
@cahe mentioned something that made sense about that:  He suggested color change on infill - that would mitigate the impact of a gap.   If that is not possible, consider that it is typical to print the perimeter from the inside-out which would also mitigate any gap.

cahe, 2017-01-16:
I think I'd be super hard to detect when infill starts, since the script relays on Z values. A gap on the inner perimeter on the other hand doesn't seem that bad.

prusa3d-bb, 2017-01-17:
Hi there Maciej,

Could you please describe that a bit more? :)

Best regards

—
Tomáš Chvalina
Customer Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
621/8a Prvniho pluku, 18600, Prague
shop.prusa3D.com

--- original message ---

--- end of original message ---

cahe, 2017-01-17:
@lbussy Can you provide us with some photos?

bubnikv, 2017-01-17:
I personally don't see this a big issue. I am using color print time to
time and after I confirm the filament change I just grab the excess thread
of material with tweezers. This way there is neither excess material nor a
gap on the printed object.

lbussy, 2017-01-17:
I'll try to get photos/video if possible.

p3ck, 2017-05-11:
I find this annoying as well.  It has added a blob of extra filament on many prints of mine.  The print head is moving while it's extruding so it's very difficult to catch sometimes.

boelle, 2017-12-04:
since @lbussy & @p3ck  have the issue but not provided photos etc. i think its safe to say that this one is more or less solved and could be closed

closed 2017-12-05 by akukan

domme81, 2019-02-06:
Same issure here. Just contacted the support because of that. After changing the color, just before the head moves back to the object it extrudes like one or two extra centimeters of filament in my case, that, if you don't try to catch it very fast, sticks somewhere to the object then...

And apart from that, a user option how much material is extruded before the "right color now?"-question would be fine. When using very different colors like black and white I often select "no" a few times to make sure the colors come perfectly clean out of the nozzle. But that would just be a nice-to-have. Thanks! :)

Enverex, 2019-03-01:
This is definitely still an issue and almost ruined one of my prints. After the filament swap, when you say OK to the change, it then squirts out 6cm or so of filament and travels to continue the print immediately (this was a low height print so actually getting to the nozzle was almost impossible).

I had to pause the print and try to pull the excess filament off the print itself (because it had already just embedded it) before letting it continue.

If you need help visualising this; Think of the filament change process, and then a split second before the head moves back to continue the print, it almost instantaneously squirts out a long thread of filament. There's no pause after this happens and it's not a slow extrude.

tracernz, 2019-03-02:
Yes, I saw that last time I did a filament change. Luckily managed to catch the string as it was travelling back to the print.

domme81, 2019-03-02:
My current ‘workaround’ is keeping the during the color change extruded material completely hanging out of the nozzle (not taking it off!), then hold it when it’s cool enough with one hand and THEN select OK to return to the print. That way the now falsely extruded filament is easier to catch because it will physically stick to the old one (that is still hanging there, which you already grabbed and hold). Might not work for everyone depending on the distance to the object the head has to travel and the amount of falsely extruded material. And a real fix would be very nice of course, but that helped me a bit for now saving some of my prints.

dsdimitrov, 2019-08-17:
Hit the same issue. Once it extrudes and asks if the color was changed successfully, I believe it should not extrude any further. What it does it extrudes some more material while moving to the printed object. This ruined two of my prints because I was not fast enough to catch it.

gogglespisano, 2020-05-17:
This is definitely still a problem and wasn't solved simply because no one provided a photo.  The comments made it clear it happens so fast and sometimes at such a low height it's very hard to get a photo.  I tried and failed.

The "pressurizing" reason for having this extra extrusion doesn't make sense if you proceed with printing immediately after the filament change.  Even after the extra extrusion, the head is still moved back over to the work area and a speed slow enough to have more filament ooze out.  Not really any different than the one second earlier when the filament change finished.

Please reopen this bug based on the continued examples and descriptions of the issue.  At the very least this should be configurable.

Area5142, 2020-05-18:
There is still a open issue in the Prusa-Firmware section: https://github.com/prusa3d/Prusa-Firmware/issues/2502
Where it is marked as a bug.