prusa3d/Original-Prusa-i3 #21: Problems with extruder models

tkircher, 2017-02-27:
The extruder body and cover models both have issues where holes don't go all the way through. This is true of both the SCAD files and the generated STLs currently in the repository. Increasing the Z-axis lengths of the appropriate cylinders in extruder-main.scad solves these problems.

Also, the bearing support shaft, which fits into the extruder idler, doesn't exist in the repository. However, it did come included with the Prusa i3 mk2 kit and is an essential part. I'd be happy to provide patches for both of these issues.

ryanneufeld, 2017-03-16:
This was done intentionally. Since you can't print circles in mid-air. Just use a 3mm drill-bit and clean out the holes

tkircher, 2017-03-18:
I had no trouble printing these on my Prusa. If you're having issues, it's better to print with a slightly tapered hole using a conical cutout than to require unnecessary post-processing. I've done that on several designs that had too much gap to bridge reliably, and cap screws still seat perfectly well.

ryanneufeld, 2017-03-18:
I am some explaining it. I am not even a contributor.

tkircher, 2017-03-18:
Normally I would just make my own fork, but the missing parts, at the very least, should get addressed here. There are other things like the reinforced Y stage motor mount that should probably also get included in the main repository.

akukan, 2017-12-05:
Explained, by design

closed 2017-12-05 by akukan