prusa3d/Original-Prusa-i3 #29: prusa super switch

JustinCaseOfWolf, 2017-05-31:
any plans to upload that?

JustinCaseOfWolf, 2017-06-01:
this one https://d17kynu4zpq5hy.cloudfront.net/igi/prusa3d/BMhgGmov5EBXZaxu.large

JustinCaseOfWolf, 2017-06-10:
anybody reading these messages?

madhunm, 2017-06-27:
+1 Any plans to upload the super switch schematic?

cscott, 2017-07-18:
Some small clues at https://github.com/MarlinFirmware/Marlin/commit/5b11b33854c9950b9a98005aa7783b1fe67b6269 FWIW.

AJMartel, 2017-09-14:
I am looking for the "MK42 stepper SWITCH rev 0.4" schematics as well.
 
[image]

JustinCaseOfWolf, 2017-09-14:
that is a very different design -fairly simple to gues swhat is going on there :)

JustinCaseOfWolf, 2017-10-27:
just to make it a bit clearer the circuit is so:
- assume A and B as being the 2 line inputs that can be either high or low
- C is A NOR B (through the 2 transistors circuit see http://www.electronics-tutorials.ws/logic/log46.gif for an example)  C will be high only when both A and B are low.
a relay has one control input, one controlled input and 2 output (normally open and normally closed). subject to control input being high or low we will see controlled input to either normally open or normally closed outputs.
basic diagram:
                     / output when control input is high
control input -- controlled input
                     \ output when control input is low

the 3 relays are defined as such:

    / M1 ( this will be power when both A and B are low)
C  - power
    \ FWD1 (this will output power when either A or B or both are high)

    / FWD2 (this will be power when A is high which makes C low)
A  - FWD1 (this will be power in when either A or B or both are high) (this is output on low from C relay)
    \ M2 (this will be power when A is low and B is high - we need C low)

    / M3 (this will be power when A and B are high which makes C low)
B  - FWD2 (this will be power in when A is high which makes C low) (this is output on high from A relay)
    \ M4 (this will be power when A is high and B is low which makes C low)

in plain english:
 -first relay will control a motor(1) when both inputs are low, otherwise will forward the current to second relay. 
 -second relay will forward current to third relay when first input is high or control a motor(2) when second input is high. if second input is low then first relay is active. and second relay doesnt have current
 -third relay will output power to a motor(3) when second input is high or to a motor(4) when second input is low. third relay only has power when first input is high, otherwise it will not give current to any motors.

hope this helps.

ok1hra, 2017-10-27:
Source now available here https://github.com/prusa3d/MK2mm_Stepper_switch

bubnikv, 2017-10-27:
Documentation has been provided. Closing this issue.

boelle, 2017-12-04:
@bubnikv i think you forgot to hit the close button, there are also these other issues that could be closed

https://github.com/prusa3d/Original-Prusa-i3/issues/32
https://github.com/prusa3d/Original-Prusa-i3/issues/30
https://github.com/prusa3d/Original-Prusa-i3/issues/28
https://github.com/prusa3d/Original-Prusa-i3/issues/21
https://github.com/prusa3d/Original-Prusa-i3/issues/18
https://github.com/prusa3d/Original-Prusa-i3/issues/17
https://github.com/prusa3d/Original-Prusa-i3/issues/13
https://github.com/prusa3d/Original-Prusa-i3/issues/10
https://github.com/prusa3d/Original-Prusa-i3/issues/9
https://github.com/prusa3d/Original-Prusa-i3/issues/6
https://github.com/prusa3d/Original-Prusa-i3/issues/1

These ones you might want to pay some attention on

https://github.com/prusa3d/Original-Prusa-i3/issues/16
https://github.com/prusa3d/Original-Prusa-i3/issues/12
https://github.com/prusa3d/Original-Prusa-i3/issues/11
https://github.com/prusa3d/Original-Prusa-i3/issues/8

bubnikv, 2017-12-04:
@boelle Thanks for your efforts. I am more on the Slic3r side of the development, it is up to @michalprusa @akukan @ok1hra to close the HW related issues.

boelle, 2017-12-04:
@bubnikv  @michalprusa @akukan @ok1hra 

no problem, since the isue list was so short i read them all and a lot of them did not make sense to keep open anymore. On a few of the issues i have made a comment on why i think they can be closed

closed 2017-12-05 by akukan

boelle, 2017-12-05:
@akukan  thanks for cleaning up a bit

i made this list of issues that could be closed and you got a few

#32
#30
#28
#21
#18
#17
#13
#10
#9
#6
#1

akukan, 2017-12-05:
@boelle thanks for help, for the rest of the issues some work needs to be done so this will take little bit more time.

boelle, 2017-12-05:
@akukan no problemo

currently i'm looking at the DXF's for the MK3, they are not 100% perfect

i have started to make comments in https://github.com/prusa3d/Original-Prusa-i3/pull/36

is that the best place leave comments for mk3 ?