prusa3d/Original-Prusa-i3 #30: Can't load any stl into slic3r or simplify

MandorSawall, 2017-06-14:
I can't load _any_ prusa mk2 or mk2s stl into slic3r 1.2.9 nor Symplify3d 3.1.1.
Slic3r "stops working" and Simplify3d says "Part appears to be extremely small and may be in inches format..." and says it's loaded but I can't see it.
The dragon from thingverse loads just fine in both. https://www.thingiverse.com/thing:600550

Moorviper, 2017-06-14:
It is only on master branch.
all other branches works fine.

tested with slic3r Symplify3d and Cura

MandorSawall, 2017-06-14:
What does that mean? For example, i can't load https://github.com/prusa3d/Original-Prusa-i3/blob/MK2S/Printed-Parts/stl/LCD-cable-clip.stl

Moorviper, 2017-06-14:
Your linked files loads fine on all slicers

As i wrote:
only master branch is not working ....

http://picpaste.com/pics/Bildschirmfoto_2017-06-14_um_10.27.33-tifHI7Ey.1497429047.png

MandorSawall, 2017-06-14:
OK, I downloaded Cura and failed to open same files.
Then i clicked on the link and used the download button and now it opens everywhere.
Previously, I rightclicked the link and clicked "save target as" and it produced stl files of different size on my disk. It seems very odd to me that i can't download like that.

Moorviper, 2017-06-14:
https://git-scm.com/book/en/v1

;-)

bardiir, 2017-07-18:
@MandorSawall you can't download like that because github does render a detail page for the file there. You can clone the git repository or you can donwload it either as a full zip download on the main page of the repository or a single file on the files detail page.

boelle, 2017-12-04:
@MandorSawall seems like you got your "issue" solved

maybe consider to hit close button????

closed 2017-12-05 by akukan