prusa3d/Original-Prusa-i3 #31: LCD_cover.stl is non-manifold

unrepentantgeek, 2017-06-19:
LCD_cover.stl is full of duplicate faces that cause Simplify3D to see it as multiple parts and slice it as such.

[image]

prusa3d-bb, 2017-06-19:
Hi there,

Never seen that one. All of these .stl works just fine at our Slic3r. What about the rest of the .stl we provide for the printed parts? Do you experience same issues with these also?

Best regards

—
Tomáš Chvalina
Technical Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
Partyzánská 188/7A, 170 00 Praha 

shop.prusa3D.com

--- original message ---

--- end of original message ---

unrepentantgeek, 2017-06-20:
The file works ok in Slic3r because it has a feature (bug?) wherein it merges bodies that are close to one another.  Simplify3D attempts to print them as separate bodies and my printer is dialed in well enough that they were really close to being multiple parts.

I don't see a SCAD file for the LCD cover in the github repo (maybe I'm just not seeing it?)  I suspect that the issue is that the model has non-overlapping bodies that are unioned together.  Unfortunately OpenSCAD generates non-manifold geometry when it encounters this.

I haven't noticed this on other parts, but I'll take a look at them and see if I see any problems.

prusa3d-bb, 2017-06-22:
Hello 
Sorry for the delay in getting back to you, I am consulting with one of the guys here and to see where the SCAD file is as I do not see it there as well.
Thank you. 

—
Shane LaHaye
Technical Support

PRUSA Research

+420 222 263 718

+421 220 570 305

621/8a Prvniho pluku, 18600, Prague

shop.prusa3D.com

--- original message ---

prusa3d-bb, 2017-06-23:
Hello
One of the guys who are more skilled than I am was able to locate the SCAD file for the LCD cover, 
Here it is 
https://github.com/prusa3d/Original-Prusa-i3/blob/MK2S/Printed-Parts/scad/LCD-cover-ORIGINAL.scad
Thank you 

—
Shane LaHaye
Technical Support

PRUSA Research

+420 222 263 718

+421 220 570 305

621/8a Prvniho pluku, 18600, Prague

shop.prusa3D.com

--- original message ---
On Thu, Jun 22, 2017 at 09:55 pm, info@prusa3d.com wrote:

Hello 

Sorry for the delay in getting back to you, I am consulting with one of the guys here and to see where the SCAD file is as I do not see it there as well.

Thank you. 

—

Shane LaHaye
Technical Support

PRUSA Research

+420 222 263 718

+421 220 570 305

621/8a Prvniho pluku, 18600, Prague

shop.prusa3D.com

> The file works ok in Slic3r because it has a feature (bug?) wherein it merges bodies that are close to one another. Simplify3D attempts to print them as separate bodies and my printer is dialed in well enough that they were really close to being multiple parts.

> I don't see a SCAD file for the LCD cover in the github repo (maybe I'm just not seeing it?) I suspect that the issue is that the model has non-overlapping bodies that are unioned together. Unfortunately OpenSCAD generates non-manifold geometry when it encounters this.

> I haven't noticed this on other parts, but I'll take a look at them and see if I see any problems.

--- end of original message ---

unrepentantgeek, 2017-06-24:
Ah I see, I was on the wrong branch (MK2)  The STL is non-broken in the MK2S branch, perhaps you could merge the files over?

Thanks for your help!

closed 2017-06-24 by unrepentantgeek