prusa3d/Original-Prusa-i3 #33: Add Option to prevent Display Auto-Reset

bardiir, 2017-07-17:
I'd like to see an option to disable the Display automatically switching back to the info screen after a few seconds or at least an option to increase the timeout.

It's pretty annoying if you're in the move axis menu for example, working on the extruder moving the axis via the menu and the next time you need to move the axis again it's back at the start screen.

prusa3d-bb, 2017-07-18:
Dear Alexander.

Thank you for your feedback.

There is a single common timeout value for all menus, with the only exception being the live adjust menu, where the timeout is extended. So we can selectively extend the timeout.

What menus do you wish to have the timeout value extended?

Thank you,

—
Vojtěch Bubník

--- original message ---

--- end of original message ---

bardiir, 2017-07-18:
Where I most definitely noticed it being annoying is the move menus when manually moving any of the axis. Those should in my opinion have very long timeouts as you most likely use them during other maintenance operations.

And overall I think you should increase the timeout by about 50%, it just feels very quick to drop out of the menu again if you look something up in the manual for example while navigating the menu.

bubnikv, 2017-07-18:
I have created a report in the firmware section:
https://github.com/prusa3d/Prusa-Firmware/issues/149
I am closing this issue, please follow the other issue. In the future, please report the firmware issues there.
Thank you.

closed 2017-07-18 by bubnikv