prusa3d/Original-Prusa-i3 #34: Z rods are not parallel nor perpendicular to XY

gregsaun, 2017-09-19:
I discovered that Z rods are not parallel nor perpendicular to XY due to a misalignment of rod hole in z-axis-bottom and z-axis-top.

In z-axis-bottom you translate the cylinder by **3** in Y (line 45) :
```scad
translate([25+4.3,3,-1]) rotate([0,0,0]) cylinder(h = 50, r=4.125, $fn=15);
```

But in z-axis-top you translate the cylinder by **3.2**  in Y (line 51) :
```scad
translate([25+4.3,3.2,-1]) rotate([0,0,0]) cylinder(h = 50, r=4.1, $fn=15);
```

The frame mounting holes are at the same position on both pieces and as well as on the frame. This means that the Y axis are closer to each other of 0.4mm on top in comparison to the bottom.

I could verify this by measuring my MK2s printer and by importing STL in Fusion 360. Here are the results in Fusion 360

z-axis-bottom compared to the most external frame mounting hole :
[image]

z-axis-top compared to the most external frame mounting hole :
[image]

Is it done on purpose or a bug?

prusa3d-bb, 2017-09-22:
Hi there,

so this seems to be more of a bug as we don't see any reason for this and this change has been done a while ago so would have to ask about it.

Best regards

—Douglas Whyte
Customer Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
188/7a Partyzanska, 17000, Prague
shop.prusa3D.com

--- original message ---

--- end of original message ---

gregsaun, 2017-09-24:
Will it be solved when first MK3 will be shipped?

prusa3d-bb, 2017-09-27:
Hi there,

i cant say for sure as i dont have this information but with the whole new Y-frame this should be solved yes on the new MK3.

Best regards

—Douglas Whyte
Customer Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
188/7a Partyzanska, 17000, Prague
shop.prusa3D.com

--- original message ---

Serge-nt, 2017-11-17:
Hello,
I've searched the reasons for this misalignement and in fact it seems that there is no problem or maybe the problem is not there.
The point is that in the Z_bottom part the motor footprint is a bit larger than an actual motor.
real motor side = 42.2mm
footprint motor = 42.5mm // in my files
So there is a gap between the motor and the Z_bottom part of 0.15mm on each side.
I'm ok it doesn't explain the 0.2mm.
I've modified a lot my files so some values may differ from the ones on the repository but I think it's a search starting point.
Another point is that you can rectify it by moving top and bottom parts (idk how much backlash is present on M3 holes on Prusa's machines) and I've not verified if the holes on the Z_frame are compenciating this...
Lastly form what I see top and bottom parts are not (or should not be) exactly aligned.

I hope my explainations are clear enough and please excuse my poor english.

Regards

gregsaun, 2017-11-19:
Hello Serge,

Your english is good to me, but my english is poor as well :-D

Thanks to have take the time to check this problem. I did the assembly of Z frame and z-axis-top/bottom in Fusion 360 and it confirms the 0.2mm. I also checked the alu-frame.dxf and it does not take this into account.

I agree, you can rectify this with the screws (margin is big enough according to scad file, but do not know in reality). This was more to raise a flag about this problem (one can hide another one :D ).

By the way, it would be nice to reduce the diameter of the m3 holes (currently 3.6mm) to 3.3mm or 3.2mm it might makes the build more accurate (and maybe easier). As those holes are printed vertically, it might be good to apply the same technic as in pushift_rod module of x-end.scad. That was just an idea and I understand that lots of parameters are going in this...

I take the opportunity to thank you for sharing all these scad files, I learned a lot by reading them! Can not wait to check new MK3 files (and to receive my MK2.5 upgrade)!

Regards

Serge-nt, 2017-11-21:
Hello,

I've double checked and yes you are totally right. I've aligned the Z-frame holes with top's and bottom's in openscad and there is a misalignement of 0.2mm. Here's the code : 
`use <z-axis-bottom.scad>
use <z-axis-top.scad>
import("FRAME_MK2.dxf");

translate([ 79.4,400.9,1]) rotate([90,-90,0]) color("blue") z_top_left();
translate([464.4,400.9,1]) rotate([90,-90,0]) color("blue")z_top_right();

translate([ 81.4,85.9,1]) rotate([90,-90,0]) z_bottom_left();
translate([464.4,85.9,1]) rotate([90,-90,0]) z_bottom_right();`

From front and back views we can see the problem.
I think the files should take more advantage of openscad's power like making holes parametric as you mentioned.
Let's hope this will be solved for the MK3 but for all versions as well ... it's not a big deal and I think DIYers will still build the older printers because of the bill less specific.

Regards.

gregsaun, 2017-11-21:
Thanks for that! I tried your code and as you mentioned we can clearly see it, nice way to test this. However, I could not find the FRAME_MK2.dxf  (only alu-frame.dxf from Josef repo : https://github.com/josefprusa/Prusa3-vanilla) and this frame does not align with z_top and bottom... Anyway, after realignment I could see it.

Hope this will be fixed for all versions as well :)

zapman42, 2018-01-03:
Hi, I got my I3 MK3 just a few days before christmas and built it right away. I like the printer very much but I am having a rattling/ vibration at the top Z-Axis mounts. No matter how I attach the top/ bottom holders, the lead screw is not parallel to the smooth rod and the lead screw can make contact with the top Z-Axis holder (details see link below). 
--> Could this problem be caused by the bug you describe here?

I also opend a thread at the "Prusa Answers" page here: http://manual.prusa3d.com/Answers/View/858/Z-Axis+screws+are+not+perfectly+vertical+and+cause+rattle-+vibrations

gregsaun, 2018-10-17:
It has been fixed on MK3 but as far as I know not on MK2s (which is still sold)