prusa3d/Original-Prusa-i3 #39: MK2 Frame not included

bwattendorf, 2017-12-06:
I was wondering what the power supply screw hole sizes and placement are on the MK2 frame? I have an older Prusa frame and was looking to mount the newer power supply.

prusa3d-bb, 2017-12-06:
Hello,

You can find more info on those over here: http://manual.prusa3d.com/Guide/7.+PSU+%26+Heatbed+assembly/297#s5043

Best regards,

Jennes De Schutter
Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

Partyzánská 188/7a, 17000, Prague

shop.prusa3D.com

--- original message ---

--- end of original message ---

bwattendorf, 2017-12-06:
> Hello,
>
> You can find more info on those over here: http://manual.prusa3d.com/
> Guide/7.+PSU+%26+Heatbed+assembly/297#s5043
>
>
> Best regards,
>
> Jennes De Schutter
> Customer Support
>
>
>
> PRUSA Research
>
> +420 222 263 718 <+420%20222%20263%20718>
>
> +421 220 570 305
>
> Partyzánská 188/7a, 17000, Prague
>
> shop.prusa3D.com
>
> --- original message ---

I am looking to know where the holes are on the frame so I can drill holes
on my frame. the holes I am referring to are on step 4 of the guide you
just provided a link to.

gregsaun, 2017-12-06:
You will find the DXF in MK2S branch here :
https://github.com/prusa3d/Original-Prusa-i3/tree/MK2S/Printed-Parts/scad/FRAME_MK2.dxf

bwattendorf, 2017-12-06:
Huge Thank you..

closed 2017-12-06 by bwattendorf