prusa3d/Original-Prusa-i3 #41: Power supply electronics base SCAD source missing

fiveangle, 2017-12-11:
Any idea when the power supply electronics base SCAD source files will be up ?  I want to get it modified for rear-exit of the NEMA-15 [like I did for the MK2/MK2S](https://www.thingiverse.com/thing:1942906), so it will be ready when my kit arrives :)

Thanks,

-=dave

prusa3d-bb, 2017-12-11:
Hi there,

I just forwarded this message to our developers. They will take a look at it :)

Best regards

—
Tomáš Chvalina
Technical Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
Partyzánská 188/7A, 170 00 Praha 

shop.prusa3D.com

--- original message ---

--- end of original message ---

erikcramerjr, 2018-01-21:
Any update on this, im still missing the files.

fiveangle, 2018-02-04:
Too busy making Extruder Motor Visualizers perhaps ? 😉

fiveangle, 2018-02-25:
Can someone provide clarification as to why source for this part is not available ?

nicw, 2018-02-25:
I'd be interested in this as well. I'm assuming there is some legal concern around releasing this because it's the power supply, or does Prusa consider this part to be their own intellectual property?

jltx1, 2018-03-09:
I am eagerly awaiting fiveangle's rear-exit modification, so please post the scad files.  Thank you.

gregsaun, 2018-03-10:
As others, I would like to use the SCAD/STL file for my [project](https://github.com/gregsaun/prusa_i3_bear_upgrade). Any news?

cnschu, 2018-04-05:
i contacted support a few eeeks ago about this issue, but other than "i'll forward it", no fix so far.

fiveangle, 2018-04-05:
3 months later and nothing.  This should have been an explicitly easy-to-fulfill request since they continue to ship this part with every printer.    With an entire quarter of a year come and gone with no action, and at least one other unanswered request for the same in #60, it is difficult to conclude anything other than this is intentionally and knowingly being ignored.  The question is why ?  If they deciced to not release just this part under GPL like the rest of the printer, why not just state so instead of ignore the community ?

Anyway, I've executed on an alternate solution and have no further need for this personally so have half a mind to close this issue, but will leave open if others want me to continue pursuit.

-=dave

lecktor, 2018-04-06:
Today, I was at Prusa Research office and asked the same... so basically in 15 minutes I've received stl files to my email. Adding files to here... (A9 version)
[PSU_cover_MK3.zip](https://github.com/prusa3d/Original-Prusa-i3/files/1885211/PSU_cover_MK3.zip)

fiveangle, 2018-04-07:
zip contains STL so now that there is proof of them distributing binaries, I guess they are now on the hook for the source (I guess like Creality3D : )

-=dave

fiveangle, 2018-04-09:
@3d-gussner - I suggest you review how the GPL works, and the license present in the repository for the MK3:

https://github.com/prusa3d/Original-Prusa-i3/blob/MK3/LICENSE

Binaries is not the same as source code.  Yes, you can modify STL just like you can try and disassemble firmware, but it is not the same as access to the source code which the GPL guarantees.

I don't even personally need this part any longer, but the spirit of open source software is being flouted, which is surprising for such a pillar of the RepRap community as Prusa Research is.  I mention all of this only out of principle and I'm surprised others here are accepting of this.

How can we expect companies such as Creality3D to honor the GPL when our most beloved and beneficial commercial venture in the community cannot do the same ?

-=dave

ManuGithubSteam, 2018-04-09:
@fiveangle You have a valid point there! I remember that i did download the source for the mk2s and there was also a folder with scad files. So at some point they released the mk2 source.

I attach the source files here:
https://drive.google.com/open?id=1R9tPsnpWZKzqlQPsWTE4ByGqfn2H1AZL

I will probably hold on to them a while longer, seen the reluctance to relase the source of the mk3.....

fiveangle, 2018-04-10:
@ManuGithubSteam - the MK2S SCAD source is published in this very repo in the MK2S branch:

https://github.com/prusa3d/Original-Prusa-i3/tree/MK2S/Printed-Parts/scad

-=dave

akukan, 2018-04-17:
Both STL and SCAD files for psu cover are published now. Sorry for terrible delay, shortly, the main reason ( same as with MK2 version during release ) was concerns regarding people working in area where they can do something wrong in 110/220 V installation.

maximlevitsky, 2018-04-17:
I must really say thanks to you, even though I won't probably be printing this cover for now.
I am really happy to see all the parts open and that is what especially makes Prusa such a great printer.
Thank you!!

fiveangle, 2018-06-28:
smh

-=dave

closed 2018-06-28 by fiveangle

fiveangle, 2022-07-20:
Oh the irony…

[image]