prusa3d/Original-Prusa-i3 #44: Mk3 nozzle

thsturgill, 2018-01-05:
[nozzle-fan-mk3_1.zip](https://github.com/prusa3d/Original-Prusa-i3/files/1607129/nozzle-fan-mk3_1.zip)
Lettering pokes into air chamber causing roughness, Increase $fn value to give smoother curves for better airflow?, increase slant in are above opening to force air in downward direction.

ManuGithubSteam, 2018-05-05:
I would like to know what you mean but I'm not that deep into fff yet. Could you simplfy what's going on?

thsturgill, 2018-05-05:
In OpenSCAD, each element is positioned separately. The positioning of the lettering is such that portions show in the inside in the nozzle air flow. Also a special variabe ($fn) controls how many segments a 'curved' surface has