prusa3d/Original-Prusa-i3 #47: Slic3r settings for spool holder

probonopd, 2018-01-08:
The new spool holder can apparently break rather easily, so people may need to print replacements.

Can you please provide the exact Slic3r settings for the `Spool-holder.stl`, e.g., print settings, number of perimeters, infill, etc. to match the original. Is the original made of PET?

Will the new spool holder only fit on the MK3 or also on the MK2S?

probonopd, 2018-01-08:
By looking at http://www.prusa3d.com/downloads/others/mk2s_upgrade_gcodes.zip with http://gcode.ws/ it seems like those were printed with .20 mm Slic3r default settings but infill "grid".

Am I right?

bubnikv, 2018-01-08:
> gcodes.zip with http://gcode.ws/ it seems like those were printed with

You can import the Slic3r config from the G-code into slic3r. This works
even if the G-code has been generated by the Prusa Control.

probonopd, 2018-01-08:
Having broken one more spool holder, I now believe the key to installing it is to carefully put pressure where the __large arrow__ is in the picture below, and only very little pressure where the small arrow is.

[image]

Can someone confirm?

probonopd, 2018-01-08:
If this proves to help, then maybe the assembly instructions could be updated using this information.

Zaf9670, 2018-02-09:
I broke mine and just printed the part with more infill and external layers. I felt it was more of an issue with the part being made too weak. I also used ABS. My new spool holder feels more indestructible.

MartinMajewski, 2018-02-09:
The problem with any original Prusa spool holder is that there is no elaborated clamping mechanism. I broke the first one also on the first try. Therefore, I came up with something, that does not require brute force for installation.

[https://www.thingiverse.com/thing:1832235](https://www.thingiverse.com/thing:1832235)

It is not sorcery to provide a path for the frame to move along during installation! Come on Prusa team; you can do better, I know it!

prusa3d-bb, 2018-02-12:
Hi there,

thank you for your message. It's possible, that you had some kind of old type of filament holder. At the moment we have a really good type of filament holder, which is tested a working very good.

If you will have some other questions, feel free to ask.

Best Regards

—Pavel Pelčák
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

https://www.thingiverse.com/thing:1832235

--- end of original message ---

brendabell, 2018-06-19:
@prusa3d-bb Granted I'm new to this party and the new type of holder is significantly better than the older one, but mine just snapped after 2 days of use and I'm currently trying to figure out how to rig something so I can print a new one. I think I'll definitely be looking at some of the other designs. The clip on the Prusa one seems to be under too much stress with a full spool.

jamesneal, 2018-06-19:
@brendabell I think this one is a really good alternative to the Prusa one: https://www.thingiverse.com/thing:2640009

* It prints in about 3 hours.
* It clips on easily and doesn't crack.
* The arm is narrower, so you can use spools with smaller holes (AMZ3D)

brendabell, 2018-06-20:
@jamesneal Interesting. Thanks for the ink.

MartinMajewski, 2018-06-20:
@brendabell Or if you like something more advanced: https://www.thingiverse.com/thing:1832235

brendabell, 2018-06-22:
Since I was dead in the water with a broken spool holder and either needed to rig something or source something to print a new one, I got one of these: https://www.amazon.com/gp/product/B01MRCV4U1/ref=od_aui_detailpages00?ie=UTF8&psc=1

It was dirt cheap and makes a decent emergency spare.