prusa3d/Original-Prusa-i3 #49: Dimension error on Z-axis-bottom-right and -left

RH-Dreambox, 2018-01-26:
Hi,
I have found a dimension error on the two parts Z-axis-bottom-right and -left that should be fixed.
The console is 0.5 mm too narrow and the Z-motors can not be mounted centered in the intended location.

The Z-screws will therefore lean relative to the Z-smooth-rod, which may cause operating problems.
Please, read a detailed description in my post "Fine tune your MK3 Z axis.

[https://shop.prusa3d.com/forum/original-prusa-i3-mk3-f30/fine-tune-your-mk3-z-axis-t13806.html](url)

gregsaun, 2018-01-26:
Thank you for finding and sharing this, I could reproduce as well on my side with MK2s and MK3. Here is a view with (motor aligned on screws) : 

[image]

stahlfabrik, 2018-08-13:
Was this issue resolved with the R2 parts?

RH-Dreambox, 2018-08-13:
No, I had to fix that on my own details.

c-stewart, 2018-08-13:
> Are there modified R2 parts that JUST include a correction for that issue?

I've only seen corrected versions that also include the extra slot for
quick leadacrew removal. Made by both of the other people posting in this
thread actually.

stahlfabrik, 2018-08-13:
Yes (but thanks!), I am aware of two version (JTLX and Gregsaun). But these are not just "pure" fixes. They change a lot and optimize for PETG. I want an official fixed (and why not also improved) version, that is PETG or ABS ready...:-)

c-stewart, 2018-08-13:
Indeed, I think you stated that clearly in your other message. I was just
answering 'no' to your question :)

c-stewart, 2018-08-13:
You could try tagging some of the developers in the message stream here to
try to get their attention and make a new part. I'm not sure who's in
charge of printed parts though.

c-stewart, 2018-08-13:
Or if someone can alter the r2 part with just the fix they could create a
pull request.

stahlfabrik, 2018-08-24:
Can somebody with the right understanding of the issue please create a Pull Request -- or any word from Prusa Research if that is a valid issue and how to patch which scad file?

triplepoint, 2018-08-24:
Just to add to the conversation, I think this i why we don't see pull requests accepted on this repository:
https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/3d-model-release-plan-and-the-parts-git-repo-in-ge-t23803.html