prusa3d/Original-Prusa-i3 #50: Improvements to the X-axis and Y-axis belt drive

RH-Dreambox, 2018-01-26:
A problem with current belt drive is that it does not run parallel to the horizontal plane.
That means two problems.
1)
When the belt is angled from the horizontal plane, the transport distance of the E unit will be slightly shorter than would be the case if the belt was completely horizontal.
This can be easily demonstrated with Pythagore's formula where the E device transports along the hypotenuse instead of the opposite katete, which would be optimal.
2)
In case of rapid movements, small strain vibrations can occur that affect the result of the printed surface.
This is because the belt must also be moved vertically up and down quickly. This does not occur if the belts are completely horizontal.

Please, see a detailed description of both my posts regarding Fine tune MK3, X- and the Y-axis.

[https://shop.prusa3d.com/forum/original-prusa-i3-mk3-f30/fine-tune-your-mk3-x-axis-t13853.html](url)
[https://shop.prusa3d.com/forum/original-prusa-i3-mk3-f30/fine-tune-your-mk3-y-axis-t13851.html](url)

prusa3d-bb, 2018-01-29:
Hello,

thank you for this information. We'll get on it as soon as possible.

Best regards,
—
Jennes de Schutter
Technical Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

https://shop.prusa3d.com/forum/original-prusa-i3-mk3-f30/fine-tune-your-mk3-x-axis-t13853.html

https://shop.prusa3d.com/forum/original-prusa-i3-mk3-f30/fine-tune-your-mk3-y-axis-t13851.html

--- end of original message ---

gregsaun, 2018-02-01:
I have also observed another problem with the x-end-idler :

[image]

In this image, the space for the screw head is bigger than the space for the nut. This means, with tension, the screw will tilt at about 1°. This is not a big angle but apparently it is enough on my printer to have the belt always on one side of the idler (bearing) and I could never have a belt properly aligned on the x-end-idler.

This is also reinforced by the screw hole diameter which is too big (3.6mm for M3 hole), this should have been reduced by moving from ABS to PETG.

You also have the exact same issue on the y-belt-holder.

Nine-X, 2018-02-04:
I did modification in scad and stl file of x-end-idler (reduce pulley screw holes) , successfully reprinted with PETG , with 200um layer/10% infil, now screw is inline, now you need to actually screw in the screw into a part.
also reduced hole for head of screw, to spread out tension on part.
please test part from pull request, and send me feedback.

mickstone, 2018-02-08:
Can this get rolled into the MK2S X-Idler too please?

triplepoint, 2018-07-05:
So did this ever get resolved?  Was there a comparable fix in the mainline models that addresses this?

DVNLO, 2018-07-19:
I'm wondering the same thing.

Nine-X, 2018-07-19:
I got printer parts replacement from prusa , and looks like they fixed production code , but i don't see this issue fixed on github code.

triplepoint, 2018-07-19:
If there are improvements to production parts that aren't being released, you could easily see that as a violation of Prusa's claim to be open source.

Someone on the engineering team needs to review the procedure for developing and releasing changes to part designs, and ensure that all development results in released design files here, at the same time they're released to production.

nospoon, 2018-10-16:
Any update on that?

OutlawAndy, 2018-10-16:
It would be great if you guys could take the time and learn how to use Github effectively.  There are very clear and easy mechanisms for referencing, and even [closing related issues via standard commit messages](https://help.github.com/articles/closing-issues-using-keywords/).
Doing so would make your jobs easier, eliminate all of the "Is this fixed yet?" comments that are surely filling up your inboxes, and greatly improve your relationship with your customers and the open source community as a whole.  Please consider.  Thanks,

  -Andy

trevjonez, 2018-10-16:
@nospoon you can view the commit history of a file to see a list of commits that have changed the file. 

https://github.com/prusa3d/Original-Prusa-i3/commits/MK3/Printed-Parts/scad/x-end-idler.scad

If we look at the commits since the date of this issue start we can see the only two commits also adjust the version text of the part label.

[x end r2 changes in this commit](https://github.com/prusa3d/Original-Prusa-i3/commit/653881f62fed4d9aa119490394b4d52f3aa85834#diff-b827aff6f4a29860c6f11514e4f0a7f2)

[x end r3 changes in this commit](https://github.com/prusa3d/Original-Prusa-i3/commit/cd66d94f6f95953803c3b885f76cc31ffb3a1f70#diff-b827aff6f4a29860c6f11514e4f0a7f2)

OutlawAndy, 2018-10-16:
or you could just include `fixed #50` in the commit message that fixed it, since you know if you just fixed it or not.  Then none of us have to go on a scavenger hunt every time we run into another issue with your printer..