prusa3d/Original-Prusa-i3 #53: extruder-body.stl is upside down in repository

mrjohnc, 2018-02-02:
The extruder-body.stl is upside down for good printing (the flat section is at the top)

Thanks

prusa3d-bb, 2018-02-02:
Hello,
Thank you for letting us know, I will forward the info further.
For now, feel free to update the orientation in Slicer.
Kind regards

—
Lukáš Vacek
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

--- end of original message ---

Patronics, 2018-02-11:
This was fixed in this commit: https://github.com/prusa3d/Original-Prusa-i3/commit/d6aa325fdc8c19dfbc4b77a18f63a8a8861af73e#diff-efaa2dff1faef925b3900f13f288df98
so this issue can now be closed.

closed 2018-02-12 by akukan