prusa3d/Original-Prusa-i3 #59: Parts labeling should be aligned with labeling of parts in sold kits

funkycode, 2018-02-19:
I got MK3 printer and I see that some if not most parts there have label. According to documentation, those labels are the file version used to print.
Parts are labeled as Ax or Bx, where x is the number. As i see in repo the labeling is Rx.
In general the purpose of it is if I buy new kit and I want to be updated to the latest version of printed parts - I would be able to determine which ones I need to print instead of reprinting everything.

funkycode, 2018-02-19:
My bad. A and B are references for Beta and Alpha

closed 2018-02-19 by funkycode

RyanTheAllmighty, 2018-02-27:
@funkycode 

So one of my parts is B3 but in this repo it's labelled as R2. Which is newer. Should I be printing off the R2 part or is my B3 part newer?