prusa3d/Original-Prusa-i3 #62: Sleep indication after first layer

malamber, 2018-03-03:
for the past week, I have been getting an indication on the LCD screen saying "Sleep..." Right after the first layer completes.  This is not interfering with the process though.

I am currently printing with PETG at 90° on the bed and 140° all the nozzle.  The fan is off on layer 1, and 50% after that.

I have included the log file
[prusa.log](https://github.com/prusa3d/Original-Prusa-i3/files/1777714/prusa.log)

I have also uploaded to YouTube video demonstrating problem
https://youtu.be/mKG9ZuC5s7I

tarikku, 2018-03-03:
First off, full disclosure - I do not know almost anything about the Prusa i3 MK3 firmware, but have read parts of the code that have to do with the issue you have mentioned and provide my interpretation.

According to line 293 of the log, the stl you are using instructs the firmware to execute a dwell command (G4) with P500 as the parameter. This instructs the firmware to pause the command queue for 1/2 of a second. The firmware sets the message to be displayed in the bottom line to "Sleep..." (MSG_DWELL). The bottom line contents are not updated too often within the firmware, and remain at "Sleep..." probably until the printing has finished, and "STOPPED." (MSG_STOPPED) is displayed.

There doesn't seem to be an empty message defined, one would clear the bottom line of the LCD once in a while.

malamber, 2018-03-03:
Interesting.  I'm using Simplify3D to generate the Gcode.  but I doubt that it has anything to do with the messages

tarikku, 2018-03-03:
Simplify3D just instructs the 3D printer to carry out a pause.
The 3D printer firmware decides to output a message to the LCD notifying the user of the pause command. The firmware needs to be adapted, so that the last displayed message is cleared.

malamber, 2018-03-03:
Right.  so ultimately this is indeed a bug in the firmware to properly display the state on the LCD

tarikku, 2018-03-06:
My point: it is a bug related to displaying the current state on the LCD. It seems that bug does not affect printing in any way.

Another thing is that this repo only contains only stl and scad files for the printed parts themselves. If you want to report this issue, this would be the correct repo: https://github.com/prusa3d/Prusa-Firmware

closed 2022-05-05 by Prusa-Support