prusa3d/Original-Prusa-i3 #64: MK3 extruder idler shaft holes are too tight

gregsaun, 2018-03-19:
During the build of my MK2.5 I had the same issue as lot of other users, impossible to put the shaft in.

I suggest to use same technique as pushfit_rod module from x-ends.scad for this hole.

prusa3d-bb, 2018-03-21:
Hello,

Sorry to hear you are also experiencing the same issue and thanks for your suggestion on the GitHub page.

Have you been able to solve it?

Regards,

—
Blas Palencia
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

Please rate our reply ...be generous :-)

--- original message ---

--- end of original message ---

gregsaun, 2018-03-21:
Yes I use a file to increase hole

prusa3d-bb, 2018-03-23:
Hello,

Very happy to hear :)

Nice weekend!

—
Blas Palencia
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

Please rate our reply ...be generous :-)

--- original message ---

gregsaun, 2018-03-23:
Hope you will fix this for other users 😉

ManuGithubSteam, 2018-05-05:
Just to let you know there is a report on the prusa build pages, that after heavy use for two months, this hole in the idler got wider and printing was not possible anymore because the pressure was too low on the filament. So I'm not sure making it wider is a good solution. But I also do not have another good one...

prusa3d-bb, 2018-05-08:
Hello,

We know about these report but it is only happening to a small quantity of customers.

Please let me know if the issue gets worse and you need our help. You can always try to print a new part.

https://www.prusa3d.com/prusa-i3-printable-parts/

Regards,

—
Blas Palencia
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

Please rate our reply ...be generous :-)

--- original message ---

Prusa-Support, 2022-05-10:
If you are still having issues with the extruder idler please contact us at info@prusa3d.com.

Marina Curzi
Prusa Research

closed 2022-05-10 by Prusa-Support