prusa3d/Original-Prusa-i3 #68: Rambo-base in MK2.5 branch is not the most recent version

subsite, 2018-04-04:
The rambo-base.stl in the MK2.5 branch (and in the STL package from the prusa website) is marked "R1", while the downloadable gcodes are "R2". Might it be that the gcodes are made from a local branch of STLs that's not yet pushed upstream? 

I'm right now in the process of printing the 2.5 parts, and I don't want to use the gcode as it has all the Rambo parts an I only need this one. 

G-code (downloaded from prusa website):
[image]

STL (downloaded from prusa website):
[image]

Thanks!

subsite, 2018-04-09:
Thanks for your reply! 

Did you mean the R2 should be there now? I still get R1 STL both from here (https://github.com/prusa3d/Original-Prusa-i3/blob/MK2.5/Printed-Parts/stl/Rambo-base.stl )  and the prusa site (https://www.prusa3d.com/prusa-i3-printable-parts/ ). Is there some other location I'm not aware of?

prusa3d-bb, 2018-04-09:
You are very welcome! Anytime :). 

You should check our website, if there aren´t updated the STL files yet, please allow us a couple of days to do it. However, the issue has been addressed and meanwhile if the Gcode is alright for you, please use it, thanks. 

S pozdravem / Best Regards / Mit freundlichen Grüßen

—
Peter Mencák
Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

188/7a Partyzánská, 17000, Prague

shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

--- end of original message ---

subsite, 2018-04-09:
Ok, sure I can use the gcode, I'll just waste a bunch of plastic since everything else is already printed and the gcode has all the rambo parts in the same file. 

I don't want to be difficult, I just think that if you buy a product where you have to print the parts yourself, you shouldn't have to doubt that the downloadable parts are maintained meticulously.

prusa3d-bb, 2018-04-10:
Hi Fredrik. 

The parts with the designation R1 are the same as with the R2, so you can use them for sure :). 

S pozdravem / Best Regards / Mit freundlichen Grüßen

—
Peter Mencák
Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

188/7a Partyzánská, 17000, Prague

shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

subsite, 2018-04-11:
> The parts with the designation R1 are the same as with the R2, so you can use them for sure :).

That statement is simply not true. For example, in the R2 (gcode) above the screw hole next to the extruder cable slot there is a groove (supposedly to make the screw easier to reach with the hex wrench), and this is missing in the R1 (STL). Now I'm not saying this particular difference is of any consequence, but it shows it's a different revision and you can't be sure it's the only change. (The part shown in the on-line assembly manual says R1 and has the groove, so maybe that's the R1 you say is the same. The shared STL, however, is not.)

I see that you don't think this is an important issue, but I think it should be a matter of principle. If you share printable files in different formats, people should be able to count on that the files are have the same revisions. 

I've been very happy with both your products and your customer service so far, and this issue won't change that. I just don't get why this small issue is so difficult to resolve.

prusa3d-bb, 2018-04-16:
Hello,

I have addressed this. The parts should be updated on Github soon.

Best regards,

—
Jennes De Schutter
Technical Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

That statement is simply not true. For example, in the R2 (gcode) above the screw hole next to the extruder cable slot there is a groove (supposedly to make the screw easier to reach with the hex wrench), and this is missing in the R1 (STL). Now I'm not saying this particular difference is of any consequence, but it shows it's a different revision and you can't be sure it's the only change. (The part shown in the on-line assembly manual says R1 and has the groove, so maybe that's the R1 you say is the same. The shared STL, however, is not.)

I see that you don't think this is an important issue, but I think it should be a matter of principle. If you share printable files in different formats, people should be able to count on that the files are have the same revisions.

I've been very happy with both your products and your customer service so far, and this issue won't change that. I just don't get why this small issue is so difficult to resolve.

—

--- end of original message ---

subsite, 2018-04-16:
That's good news! I was very close to eating all the MK2.5 gummy bears out of frustration ;)

Cheers

subsite, 2018-04-24:
Hello, any news regarding this? The parts don't seem to be updated yet. Thanks!

> prusa3d-bb commented 8 days ago    
> 
> Hello, I have addressed this. The parts should be updated on Github soon. 
> 
> Best regards, — Jennes De Schutter
>

clearbucketLabs, 2018-05-05:
> Hello, any news regarding this? The parts don't seem to be updated yet. Thanks!

I don't think this guy can do it, he just tells someone else who then proceeds to do nothing.

sl1pkn07, 2024-05-08:
Hi

2024 and the rambo base stl is still the old version in the printables page. also here in github https://github.com/prusa3d/Original-Prusa-i3/blob/MK2.5/Printed-Parts/stl/Rambo-base.stl

anyone can fix it? print from gcode is not a option for me

greetings