prusa3d/Original-Prusa-i3 #69: MK2.5/MK3 hotend is very difficult to remove for maintenance

gregsaun, 2018-04-07:
On MK2.5/MK3 extruder, it is very hard to remove hotend if you need to do some maintenance on the PTFE tube (for example). This was much easier on MK2s extruder

Panayiotis-git, 2018-04-07:
For the MK2, I've split the extruder body in two halves:
[image]
[image]

May be something similar is possible with the 2.5/3 extruder.

matthew-humphrey, 2018-06-29:
+1. I literally had to cut the PTFE tube (and unfortunately the retention collet) to get it out.

stephanpark, 2018-06-30:
There is a trick to it.
1. Tilt the hotend assembly slightly enough to get a spudger between heatsink and Extruder body.
2. Tilt the hotend assembly back down putting the pull force on the wedge.  This keep the collar parallel and avoids pinch jamming and bend the part that can, the PTFE liner.
3. Lever the wedge while again keeping the hotend assembly parallel as possible slightly bending the PTFE tube into an S-like shape slipping the hotend clear of the capture collar.
If you did it right, it will slip right out without permanently bending PTFE or damaging anything.

You can alternately design the Extruder body's collar with a rotational clearance rather than the simple and ineffective chamfer of the original design but once you the the trick, you will no longer need to.

Mochanic, 2018-07-08:
OMG, that worked so well!  I was about to start cutting $*#&  What a stupid design that was... Definitely like the spit design much better!

amirsamad, 2018-10-28:
> There is a trick to it.
> 
> 1. Tilt the hotend assembly slightly enough to get a spudger between heatsink and Extruder body.
> 2. Tilt the hotend assembly back down putting the pull force on the wedge.  This keeps the collar parallel and avoids pinch jamming and bending the part that can, the PTFE liner.
> 3. Lever the wedge while again keeping the hotend assembly parallel as possible slightly bending the PTFE tube into an S-like shape slipping the hotend clear of the capture collar.
>    If you did it right, it will slip right out without permanently bending PTFE or damaging anything.
> 
> You can alternately design the Extruder body's collar with a rotational clearance rather than the simple and ineffective chamfer of the original design but once you try the trick, you will no longer feel the neeed to.

Yes indeed, this works. Took me a while to figure it out, but works!

Prusa-Support, 2022-05-07:
Hello! Closing this issue as it has been resolved, in the meantime we have better manuals and guides to access hotend and their parts. Sorry for the response on this from my side being only now! 

Alessandro Pantaleo
Prusa Research

closed 2022-05-07 by Prusa-Support