prusa3d/Original-Prusa-i3 #73: Spool holder version and Scad file

ChrisPetit, 2018-05-18:
Hello,

the spool holder stl here: https://www.prusa3d.com/prusa-i3-printable-parts/ is a more recent version than the stl file in the mk3 branch. Also, i can't find the scad file for the spool holder anywhere...am I missing something...thanks.

DominikPalo, 2018-08-17:
I'm also looking for the MK3 spool holder scad file, why is only the STL file in the repository?

kyjak, 2018-09-23:
Also unable to find the scad file. Since 18 may  long time passed and no results.

hillct, 2018-09-29:
Yet another bump for this issue...

It would be very helpful to get the SCAD file for this holder.

draeath, 2019-05-04:
The file appears to have been exported from [Rhinoceros](https://www.rhino3d.com), unlike the other parts (which seem to be exported by various Open Source utilities. It may be that the source file for the spool holder is not in an acceptably open format.

Jackjan4, 2021-04-29:
Bump this issue. I know that that the B4 spool holder is kind of deprecated, however it is still good usable for custom spool holder solutions.

Please update the Spool Holder in this repository to the B4 spool holder (it is still the B3 one) and add STEP or SCAD files for it.