prusa3d/Original-Prusa-i3 #76: Y-Axis Crash Detection Inconsistancies

TheZeroBeast, 2018-05-30:
#### Printer Setup
 * MK3 on 3.2.1 FW
 * Running latest R2 Printed Parts
 * Slic3r 1.40.0-Beta

#### The Issue
I have had continuous issues with Y-Axis on crash detection being offset by different amounts each crash.
This results in a small layer shift in print rendering the print a failure.

As the **Y-Axis** is using tmc2130 sensorless homing it relies on hitting the first bearing which is an offset load causing a random amount of skew from bearings and the offset load.

#### The Solution
I have redesigned the **Y-Motor-holder** to incorporate a hard stop for the hotbed belt mount so the load is centered to the motor and removes the random skew introduced.   _**PR #74 #75**_

#### The Testing
This has been tested on my MK3 with single wall cubes with flawless results as per attached photo. Extrusion issues seen are at crash/recovery point which could be improved in FW side.
[image]
[image]

3d-gussner, 2018-05-31:
@TheZeroBeast Nicely done! 
As the ball bearings are not really fixed by the U-bolts and can move a bit when hit the y-rod-holder it makes sense to use the reliable MK2/s design where the y-belt-holder hits the y-motor-holder.

BTW: The STL seams to be broken.

hst68, 2018-05-31:
can confirm it: The STL seams to be broken.

TheZeroBeast, 2018-05-31:
Ok @hst68 @3d-gussner all good now, thank you for the heads up. 👍😂
Rob

3d-gussner, 2018-05-31:
STL is fixed

hst68, 2018-05-31:
now I can't find the file....

jltx1, 2018-06-02:
STL is mirror image for some reason.  Can you fix or is base SCAD that way?

While you are at it, why not extend a tab off the new extension to add a third screw hole to tighten up the Y motor?  It's a touch flexy as is.

3d-gussner, 2018-06-02:
@jltx1 Nice idea,  as you said it had to be bit thinner as there is around 6mm space between motor and y-belt holder.

gregsaun, 2018-06-03:
Thank you for this finding and for sharing it!

3d-gussner, 2018-06-03:
@jltx1 As the third screw plastic just needs to be 1mm thinner i think that the M3x10 should fit.

gregsaun, 2018-06-04:
@TheZeroBeast I have done some measurements and you are using a length of 28.1mm from top motor hole to end of your added end stop. This is a good value if bearing is almost centered. However, if bearing is not centered and pushed in direction of the Y motor, length might become more 28.9-29mm.

jltx1, 2018-06-04:
I noticed the same.  I decided to leave it so as not to reduce Y range.  IMO, better to update assembly instructions to have folks check that the center hits first and adjust bearing as needed. 

-- Jon

TheZeroBeast, 2018-06-04:
Hi @gregsaun, I designed this to limit the loss of y-axis range and agree with @jltx1 that if incorporated build instructions should be updating to accomadate.

On a side note I have done half a dozon 6h test prints to heights over 85mm without and Y-Axis crashes detected (confirmed it is still turned on)
Previously Y-Axis crashes would happen all the time for apparently no reason and with the feature off no steps are lost. Last time I looked my all time crash count was over 400 for Y and under 30 for X.

I feel that as we retuned the belt tention/crash threshold after the installation of this part it has come to a more accurate value resulting in more robust functioning.

I'd be interested to hear how you have found your prints since this upgrade @jltx1

alientek, 2018-06-20:
where can i download this?

VVlasy, 2020-02-03:
This has died down but I still feel this is an issue.

I get at least one to two crashes on Y axis when printing a benchy.

I rebuilt and readjusted tension on both the Y axis and X axis.
I also carefully tightened the u bolts just to the point when they are not loose. I cannot feel any resistance moving either axis back and forth. even with the belts both on and off. 

Currently the values for belt state is X 283 and Y 266.

i3 MK3S kit, FW 3.8.1

Prusa-Support, 2022-06-09:
Thank you for sharing your idea.

The modification suggested in this issue could reduce the bounce across the right side of the Y-carriage: Y-axis homes once the rear bearing on the left side hits the end of the axis, and the one-bearing on the right side may have slight backlash instead.
I personally like the suggested design improvement, nonetheless, such bounce may be negligible or not concerning. In fact, the "end-stop linear-bearing" will remain a firm point of reference for the Y-homing.

Moreover, I'd like to highlight that in case of numerous crash detections, the main concern should move to that aspect in the first place, and possibly be sorted out with proper troubleshooting.

This doesn't mean that our developers can't work for a more reliable Y-homing 🙂 but, so far, the "bounce" is not a confirmed issue and this issue could be closed as stale, but please share your thoughts and let us know if the problem persists.

Please consider contacting the Technical Support before posting about new similar problems.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research