prusa3d/Original-Prusa-i3 #78: MK3 Stops in the middle of prints for a filament change. I didn't tell the print to have a filament change.

derekaward, 2018-06-03:
Firmware: 3.2.1-576

It started when a print I downloaded a file from thingiverse.com [https://www.thingiverse.com/make:496393].

Imported it to prusa control. Then started printing, it wanted to do a filament change after the second layer. This was the first time it happened to me so I didn't cancel the print and I went through the process of changing the filament. It did a filament change 7 times!

I thought it might have been the file but no I looked and the gcode and it didn't have M600 to start the process. 

I finished the print and started another one... Didn't have to change filament during that print. 

I print a few more things and none of them needed a filament change. 

Today i'm printing a vase, i'm attaching the gcode [here](https://drive.google.com/open?id=1slHa2pS3rzEfpV6WC2nNYZJPOUz-X2gf). (Using Slic3r)

This time I had to go thought the process twice. I posted on the Pursa Owner FB page and someone told me to turn off the filament sensor. I did and the print right now is still going with out stopping.

I really hope this is an oversight on my part.

derekaward, 2018-06-03:
I found the code that was getting injected. The artic

derekaward, 2018-06-03:
Keeping it open just in case.

tracernz, 2018-06-03:
This happens when the filament sensor doesn't detect the filament moving. Presumably the idea is to unload the filament, cut the (jamming) end off, and put it back in to continue. Unfortunately I have also had a few false positives from this.

derekaward, 2018-06-03:
Thanks for the feedback! Before it went to bed last night I read somewhere that a person was having the same issue and it was because they were using transparent filament. I was using transparent filament both times this accrued. I hope there will be an update to change the wording, so it doesn't sound like you mess something up in the slicer.

kris82, 2018-07-02:
I've experienced a similar issue few times. If everything goes fine after that it is just annoying but it can happen that this is actually the end of your printing. So far (I just have a printer for a week now) the issue had the following consequences: 

First when that has happened (I guess as my wife heard the same "beeps") I was away and couldn't do nothing. When I get back I've noticed information on LCD stating something about timeout and temperature (idle for too long while extruder and bed heated?)
 After this error / information there was no possibility to resume print so 20 hours of printing time was lost... I was printing https://www.thingiverse.com/thing:2901274

It has happened again today for me after printing for around half an hour of simple model. I've unloaded PLA and loaded it again but temperatures were too low when it has started to print so again printing went directly to bin.. 

I've restarted the same code and this time everything was fine up to the end without asking for a fillament change. 

In both of these cases I was using red PLA on MK3 with firmware 3.2.1-576

prusa3d-bb, 2018-07-03:
Hello,

I believe the filament sensor is giving false lectures and stopping the printer. It happens when the filament has bright or transparent colors. There are 2 things you can do:

- Disable the filament sensor while printing this materials.

- Download and install the latest Firmware just released yesterday and that improve the sensor performance.

https://www.prusa3d.com/drivers/

Can you please provide me the order number of your kit/printer?

—
Blas Palencia
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

Please rate our reply ...be generous :-)

--- original message ---

--- end of original message ---

kris82, 2018-07-03:
Hi, 

You are absolutely right, I've spent yesterday some time to analyse the source code and I can't see any any other possibility than getting a false positives from sensor. 

Currently I've disabled the sensor before some better solution will be available. I'll try a new firmware tomorrow when another of my long print will finish and new trapezoid nut arrive. 

Generally despite of possible issue with sensor I think it would be great to add an option to resume printing after heating is stopped for security. 

 I'll take a closer look at code today as it was quite late yesterday 

My order number is: 182132700.

prusa3d-bb, 2018-07-09:
Hello,

Yes, let me know how is going. Also, thanks for feedback and I will add your suggestion to our 'customer wishes' list.

Regards,

—
Blas Palencia
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

Please rate our reply ...be generous :-)

--- original message ---

dpmurphy, 2018-08-27:
I received my mk3 a week or so ago. I’ve started printing with esun petg and I see this fail at least every 30 mins. Half the prints are failures....and this is with latest downloaded today firmware.

prusa3d-bb, 2018-08-28:
Hello Daniel.

We are sorry to hear that.

Can you please me send me pictures of the failed printed models and describe the problems better? Which model where you printing? Do you have the same problem with all filaments?

Regards,

—
Blas Palencia
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

Please rate our reply ...be generous :-)

--- original message ---

dpmurphy, 2018-08-28:
Hi,
I have no issues with PLA.

The issue is filament pull….very frequent stops, unload and reload filament. If I miss the beeps and take sometime to do the unload the temperature has dropped and I get issues with the later stages not laminating correctly with the lower layers.

Attached is one of the gcode that fails, but in reality they all have issues.

I have switched off the filament  sensor as recommended in the thread and I’m 2 hours into a print without any issues. I didn’t believe I would have gotten this far with the filament sensor on.

Regards.

Dan.

OutlawAndy, 2018-08-29:
@dpmurphy You've got it, it's the filament sensor.  It happens because it cannot distinguish between filament and the extruder body if they share similar refraction characteristics.  You should be able to turn the sensor back on once you're done with this particular spool.  Good luck and happy printing!

jltx1, 2018-08-29:
Another option is to use my indirect mod which makes the laser filament sensor work with any color filament, even clear.  Right now it only works with https://www.thingiverse.com/thing:2845416 <https://www.thingiverse.com/thing:2845416> but I am looking to add it to the stock extruder if I ever get some free time.

prusa3d-bb, 2018-08-30:
> Which filament color or type are you using? If transparent or bright it can be the cause of this problem.

And can you confirm that the problem is only happening with certain colors/spools?

We are working in solved most of this problem in future firmware releases. Even in a different type of filament sensor for avoid the problems caused by the color or type of filament. But is a little bit early to speak about this possible solution.

—
Blas Palencia
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

Please rate our reply ...be generous :-)

--- original message ---

I’m using ESUN Petg Black

—

--- end of original message ---

TigerBouli, 2018-09-11:
Yes, I can also confirm: black PETG makes the problem. I started printing MK3 new extruder parts today and for that I purchased Spectrum PETG filament, as advised: black. First filament change after 2 layers, then one after another in a very few minutes (3-5 minutes). I was able to pause the print, disable filament sensor and now it is printing for 2 hours without any problems. Earlier I had only one such problem also with PETG dark blue.

prusa3d-bb, 2018-10-12:
> Hi all,
>
> I just finish to upgrade my MK2 to MK2.5. I test yesterday night some small pieces and have had no problem. After that i have launched a print for 6 hours with PLA color Rose. After 30 minutes, the printer bip and tell me to change the filament. (it was not planified) OK, i change for a first time, i put again the same filament and the print continue. After 10 minutes again, bip again to change filament. and again after 5 minutes... I stopped the print because i will not do that every 5-10 minutes.
>
> Is it the solution to switch off the filament sensor ? If it's the case, what is the benefit to have a filament sensor ?
>
> For the moment i didn't test with other color...
>
> Thank you for your help
>
> Best Regards
>
> Patrick Sierro

Hello,

It may be the problem with sensor. Filament which is used by you is bit shiny and our sensor is having sometimes problem with such filament.
We are working on this and it will be fixed with one of next firmware.
I will suggest to disable sensor while printing with this filament.

Regards

—
Bart Skonieczka

Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

188/7a Partyzánská, 17000, Prague

shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

Marius37, 2018-10-12:
Thank you for your answer and this informations. Ok, i will try to disable sensor for this filament.
I had buyed this filament on the official shop Prusa "https://shop.prusa3d.com".
Here is the link of the filament :
https://shop.prusa3d.com/en/filament/45-pink-pla-filament-1kg.html?search_query=FLM-PLA-175-PNK&results=220

Is this filament shiny according to you

Thanks for your reply.

Best Regards

Patrick

harrysamuel, 2018-11-08:
Ok, first I have not read the entire thread, I like to figure out fixes on my own. I also find PRUSA support is guessing most of the time, so why both chatting for help. I do sometimes listen to the suggestions, but a 3D printer is a mass of electronics, programing and physical parts, and no one really knows why something happened.

Here is my fix for the random pauses. I use XYZ software to assemble the 6 STL files that fit on my bed, I use Cura to create a Gcode file, and the entire thing prints. Ok it worked this once anyway. But I run my printer 24 x 7, So if it works I will keep doing it that way.

Is it the machine, no I doubt it, Firmware and software have a problem, will some bright engineer change a line of code somewhere and the problem go away, yes, will the change be intentional or completely accidental we will never know. For now my 12 hour prints run on there own, and this works for me. GOOD LUCK

Oh one more thing the AUTOLOAD filament has stopped working. I am testing the multiple solutions from Google. I think I figured out a fix by combining 3 or 4 answers into what will work. Might post the answer, once it works for me.

ManuGithubSteam, 2018-11-15:
> Today I have encountered the same problems as described above. I have the
> Prusa i3 MK3 for about 2 weeks now and not have had an issue. Since
> yesterday I started printing parts using Black PETG Filament and have had
> the printer stop in the middle of the print asking to change the filament.
> I use the Slic3r PE software and have NOT enable multiple color/material
> per print. Looking in the G-code, I can not find any M600 command. It
> happened at a layer height of 2.60mm and have no clue why I did that.
> As far as I can see the error is due to the filament being too identical
> to the original plastic of the extruder parts near the filament sensor.
> This results in sensor detecting filament error (which is not the case).

Hi,

Black Filament ist to glossy. The Devs know about this and the newest fw
should improve the detection.

But I just disable the filament sensor while printing very glossy
filament...

Hope this helps..

harrysamuel, 2018-12-18:
Just face it the Prusa I3 Mk3 was a mistake.  Wish I bought another cheap XYZ printer, never should have sold my first 3 printers. over 4,000 hours and no pause, pause, pause, pause. They ran all night every night.

JohnCoker, 2019-01-10:
I see this same problem with Prusa white PLA and firmware 3.5.1. Turning off the filament sensor allowed me to get through the print without stopping.

LolHens, 2019-01-10:
I've always had this problem with TPU, so I applied this small modification to my i3 and now it works perfectly:
https://www.thingiverse.com/thing:3129921
No pauses no matter what filament.

crashtest22, 2019-01-19:
Have had pausing issue today and was with a new Filament so I did some searching...I am having pause with sainsmart TPU in black...no other TPU has done this and its the only black filament I have

nahuelcarou, 2019-07-25:
@prusa3d-bb , I have also the same problem "stops in the middle of the print and asks for a filament change". Using Transparent PLA from OWL. 

Order Reference from my Prusa MK3s KIT: 192304150

I hope I get some positive feedback from Prusa. Not just deactivating the sensor. 

Thanks in advance, 

Nahuel Carou
nahuel.carou@gmail.com

aanelli, 2019-09-24:
Kind of a weird thing, but I'm getting this with my ender 3 pro, which doesn't even have a filament sensor

darBis, 2019-09-24:
@prusa3d-bb if you have MK3S, this exact reported issue is not same as yours. MK3S has a different type of filament sensor, that is not sensitive to filament properties. If you have a first variant of the sensor, a "lever with a tower poking up", you could try readjusting the filament sensor according to a manual that arrived with your printer. Try if it solves the issue. It is a little bit fiddly setup and needs to be adjusted correctly.

ldboehm, 2020-01-19:
This happened to me today.
It is work with a semitransparent filament...believe it or not...it is.
It is definitely tied to the filament sensor and/or autoloading.  
When I turn those off the issue goes away.

nahuelcarou, 2020-01-19:
Hi, try to calibrate better your filament sensor. If not, try to modify the mechanism for a better one. For example the one with comes with the mmu2 is more robust than the original one withouth the mmu2.

OutlawAndy, 2020-01-19:
This issue is resolved by the **mk3s** extruder upgrade.  Therefore, it should probably be closed.

On a side note, Github is really designed to handle this sort of thing automatically.
*i.e. auto closing resolved issues via commit message reference.*

*e.g. commit message that would automatically link to, and close this issue:*

    Fixed (#78)

Besides just the obvious time savings for the repo maintainers, using Github’s built in issue handling provides users documentation and a link trail they can follow to help them understand how an issue was resolved and what they need to do in order to benefit from the fix.

kic883, 2020-02-06:
I have assembled my MK3S since last week and I encoutered this issue this morning in the middle of a 10 hours print. This is not my first print (probably 10th) but this is the first time I use non Prusa brand PLA.  At first I thought I add color change by mistake, but I check there is no M600 commend in my gcode. 

The PLA that I was using is a rainbow color PLA, not sure whether this will affect the filament sensor.

cbr600, 2020-04-13:
I just started having this on my MK2S. It is the first time using Black ABS and it happens always after printing the nozzle clean line after Z calibration at the beginning of a print. I was also just starting to use the latest firmware and Slic3r. I went to Slic3r 2.2.0+ to eliminate the wait for hours for preview that I get from PrusaControl. But I was also getting problem feeding.  The extruder was indeed empty. Going to back to PrusaControl fixed that problem. I’m 18 hours into a 26 hour print without issue.

3d-gussner, 2020-04-14:
@cbr600 I am not sure if this is the correct repository to report your issue.
Maybe https://github.com/prusa3d/Prusa-Firmware or https://github.com/prusa3d/PrusaSlicer are a better place for that.

3d-gussner, 2024-01-04:
@daddyfixit Please reopen you issue in the Prusa-Firmware repository https://github.com/prusa3d/Prusa-Firmware as this is for the parts.

Tie-fighter, 2025-07-04:
I am also affected by this on my MK3S 😭
I am unsure if it is the filament sensor, but either way, the on-screen information is misleading, as it states M600 as the reason.
Slicer and Firmware are up to date.

closed 2025-08-02 by Prusa-Support