prusa3d/Original-Prusa-i3 #80: MK3 Smooth rod sizes

unixb0y, 2018-06-20:
I would really appreciate a little PDF similar to the one for the Aluminium extrusions with official dimensions for the smooth rods that are used in the MK3.
Can’t seem to find official information as the instructions don’t state the lengths. 
Thank you in advance !

Panayiotis-git, 2018-06-20:
Check the assembly guides.

unixb0y, 2018-06-20:
@Panayiotis-git you’re right, the current version does include lengths for X and Y, but the Z axis rod length isn’t specified.

Panayiotis-git, 2018-06-20:
You are right.
Check there. I've checked earlier and they are at correct lengths:
http://zaribo.org/smooth-rods/54-original-prusa-mk3mk2s-8mm-smooth-rod-kit-by-misumi.html

unixb0y, 2018-06-20:
Thanks! 2x320 2x330 2x370 👌🏼

closed 2018-06-20 by unixb0y