prusa3d/Original-Prusa-i3 #81: MK3 heat break has step that accumulates filament and causes jams

matthew-humphrey, 2018-06-29:
The Prusa-designed E3D heat break on i3 MK3 printers has a "step" inside the tube. This step extrusion problems including jams and broken filament. 

See the attached photo for a picture from a microscope of my heat break. The area in focus is some Prusa translucent green PETG that has accumulated in the step. This contamination was causing jams and poor print quality. I was unable to remove it even with repeated cold pulls.

[image]

Crunch69, 2018-06-29:
that explains the jams I keep having isnt the e3d heatbreak suppose to be smooth that is what they advertise

Zaf9670, 2018-06-29:
I have my heat break apart at home. I will take a look to see if mine has the same knot.

c-stewart, 2018-06-29:
Correct me if I'm wrong but I believe that Prusa gets custom heartbreaks
from E3D that include this difference, to be compatible with the multi
material upgrade.

murphy2712, 2018-06-29:
I believe it was only necessary for MMU v1, which is retired now.

c-stewart, 2018-06-29:
Right, but it wasn't when they started shipping mk3's. So the question is,
is this the same thing or something different?

Zaf9670, 2018-06-30:
I did not notice the step/knot in my heatbreak but I may have needed more light. I just got done replacing my heatbreak with the E3D Titanium.

matthew-humphrey, 2018-06-30:
It looks like the heatbreaks they are shipping now no longer have the problem. Many of us who got printers earlier do have this issue, though.

Crunch69, 2018-07-01:
when did prusa switch back to the stock e3d heat break?

matthew-humphrey, 2018-07-01:
I don't know if it's stock or just correctly machined not to have that step. I can confirm that replacing the Prusa HB with a genuine E3D resolved my issue.

ThaliaFromPrussia, 2018-08-04:
At which height is the step? Where the PTFE tube ends in the heat sink or where the heatsink and the nozzle come together?
[image]

matthew-humphrey, 2018-08-04:
It's actually right in the middle of the heatbreak. micro-swiss heatbreak has one as well, by the way.

closed 2018-08-04 by matthew-humphrey