prusa3d/Original-Prusa-i3 #84: SCAD files missing for new MK3 extruder improvements

matthew-humphrey, 2018-07-17:
Prusa recently posted some updates to the MK3 extruder designed to improve hotend cooling and print cooling. There are STL files posted, but no equivalent SCAD files.

The STL files are:

1. Printed-Parts/stl/extruder-body-v2.stl
2. Printed-Parts/stl/extruder-cover-v2.stl
3. Printed-Parts/stl/nozzle-fan-45deg.stl
4. Printed-Parts/stl/nozzle-fan-45deg-support.stl

Skywatchr, 2018-07-17:
STEPs would be even nicer :-)

matthew-humphrey, 2018-07-18:
Sounds like maybe Prusa is moving away from OpenSCAD (yay!). If so, it would be great to get the source files in whatever format they are in.

c-stewart, 2018-07-19:
> I agree also with having the native files. There are many improvements for
> the Coanda Effect that can be implemented!

The 2 new non binary STL's state that they're still made in openscad at
least. (On the first and last line)

jiri-jirus, 2018-09-17:
step files are now present. :(

jiri-jirus, 2018-10-24:
hey, @josefprusa can you comment on this?

matthew-humphrey, 2018-10-24:
I'm closing this since, as pointed out in the replies above, the STEP files are now available. 

As someone who spent several hours working around bugs in the Freecad openscad import just so I could convert one part (Einsy-base), I would be thrilled if Prusa started using Solid Works or Fusion 360 or some other cad tool with a more portable format.

closed 2018-10-24 by matthew-humphrey

infosisio, 2020-05-08:
Very sad turnaround for Prusa to abandon OpenSCAD and go for a proprietary solution.

infosisio, 2020-05-09:
@josefprusa You are an open project, how can you use proprietary software for the files?