prusa3d/Original-Prusa-i3 #85: MK3 45 support and shroud files missing

Psyc0Flyer, 2018-07-17:
When you click on the link it takes you to preview but does not have download button like the other new files.

nigeljohnson73, 2018-07-18:
Same here. If you click the RAW button you can then "Save as..." though.

tracernz, 2018-07-18:
It's because the fan parts are ASCII STLs, whereas the other parts are binary STLs, and Github offers the download button for binary files:
```
$ file ./*
./extruder-body-v2.stl:         data
./extruder-cover-v2.stl:        DIY-Thermocam raw data (Lepton 3.x), scale 0-43586, spot sensor temperature -33.725578, color scheme 222, calibration: offset 0.000046, slope 447404932878424866816.000000
./nozzle-fan-45deg.stl:         ASCII text, with CRLF line terminators
./nozzle-fan-45deg-support.stl: ASCII text, with CRLF line terminators
```
You can either save the raw data as per @nigeljohnson73, or checkout/clone them with git as normal.