prusa3d/Original-Prusa-i3 #86: Replacing the extruder assembly

allenfr, 2018-07-18:
Nice comments and videos for removing a heatsink extruder assembly. That went fine.  I will figure out how to remove the tubing next. Any videos on putting the assembly back in ?  I bought mine assembled, so I feel like a rocket scientist at this point !

stahlfabrik, 2018-07-19:
Hi, here is all you need https://manual.prusa3d.com

prusa3d-bb, 2018-07-20:
Hello,

You can check the assembly manuals (https://manual.prusa3d.com/), and the E3D V6 assembly guide (https://e3d-online.dozuki.com/Guide/V6+Assembly/6).

Best regards,

—
Marina Curzi
Customer Support 

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

--- end of original message ---

closed 2022-05-07 by Prusa-Support