prusa3d/Original-Prusa-i3 #88: Recommend slot for captured nuts

RetireeJay46, 2018-07-20:
When I replaced my original extruder with the latest mod, I had to re-use the nuts and bolts.  But it was impossible to remove the captured square nuts without cutting the original parts.  This led me to the realization that there is no reason for the original design to have the cover over the nut completely closed.  A slotted design would be better both for original installation (making it easier to verify correct location) and also allow for removal of the square nut at end of life for the printed part.  The design I created for this demo is not quite tight enough to hold the nut with a little friction, but you get the idea.
[image]
[OpenSlotNutCapture.zip](https://github.com/prusa3d/Original-Prusa-i3/files/2215705/OpenSlotNutCapture.zip)

tracernz, 2018-07-20:
I was thinking about that while doing the R3 extruder/fan upgrade last night. I like your idea. :+1: 

That said, I did manage to get the old nuts out with a small driver:
- put the driver into the nut, so the end of the driver is not protuding past the nut,
- press/lever down until the top side of the nut becomes visible in the hole,
- wedge the end of the driver between the top of the nut and top edge of the hole,
- push the nut out far enough so the hole through it protrudes outside the plastic part,
- pull/lever the nut the rest of the way out.

Panayiotis-git, 2018-07-21:
Try to drill a small hole at the opposite side. Then insert a small metal pin and push it out.

RetireeJay46, 2018-07-21:
I found a way - OK?  But what the post is about is a suggested improvement to the design.

ManuGithubSteam, 2018-09-06:
i second this. Getting the nuts back is a b****