prusa3d/Original-Prusa-i3 #89: No scad files for new Mk3 extruder

anraf1001, 2018-07-22:
Where are scad files of new extruder for Mk3 and Mk2.5? :D

tracernz, 2018-07-22:
Discussion: https://github.com/prusa3d/Original-Prusa-i3/commit/ad994c49ac9211c01122046ce410f98257c35ddf#commitcomment-29741639

prusa3d-bb, 2018-07-23:
> Discussion: ad994c4#commitcomment-29741639
>
>
>
> --- end of original message ---

Hello,

the new extruder is not designed in SCAD, unfortunately, so there will be no SCAD files. However, we will publish STEP files for the extruder that can be modified in most design programs. They should be posted to GitHub sometime this week.

Best regards,

Marek Míček

Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

--- original message ---

anraf1001, 2018-07-23:
It's even better. Thanks :D

Olefff, 2018-07-24:
STEP files would be awesome please. So fed up of having to convert STL's to STEP's so I can use them.
Thank you :)

3d-gussner, 2018-07-31:
@prusa3d-bb any update on the STEP files?

stahlfabrik, 2018-08-01:
They were just released

GMagician, 2018-08-09:
Just a question, importing extruder-body step file into onshape will not generate a part, only surfaces. Is it possible that such file is non manifold?

lf-, 2018-09-01:
@GMagician it's possible (likely?) that onshape is buggy :) I hit a similar non-manifold import bug with SAT files from Revit. But it's confusing because I could have sworn I saw some Prusa Research engineers working in what looked like something which was definitely not Fusion and looked kind of like onshape, so one would expect that it should import correctly.

GMagician, 2018-09-01:
@lf- all is possible. I imported correctly extruder cover while body gives me only surfaces. Someone else in thingiverse confirmed that is non manifold (and he uses Solidworks)

TNDavid, 2018-12-15:
I cannot find the step files for the new extruder, where are they?

prusa3d-bb, 2018-12-15:
Hello,

does that mean you found them?

Regards,

Marek Míček

Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

--- original message ---

@TNDavid https://github.com/prusa3d/Original-Prusa-i3/tree/MK3/Printed-Parts/STEP

TNDavid, 2018-12-16:
Yes, found them!

closed 2020-05-09 by anraf1001