prusa3d/Original-Prusa-i3 #90: How to print parts? Infill 20% or 40% - Grid or Recilinear

3d-gussner, 2018-07-31:
On the https://www.prusa3d.com/prusa-i3-printable-parts/#_ga=2.34561575.76702666.1532940068-779585920.1505648052 it says we should use following print settings:
0.2 mm layer height, infill 20% GRID and material PETG or ABS

On the other hand the 'Extruder_dyson_angled_fan.zip.amf' shows that the
- 'Extruder_dyson-body.stl' and 'Extruder_dyson_cover.stl' have:
**0.2 mm lyaer height, infill 40% Rectilinear** settings
- The nozzle_fan_support.stl and nozzle_fan_anglee.stl  have modified settings:
Fill density: 20% and Fill pattern: Grid

Also the provided 'Extruder_dyson_angled_fan_PET_6H_20M.gcode' preview shows this pattern!

Other gcodes like 'MK3_Black_parts_PETG_18H_V2_notsoldered.gcode' use
0.2 mm layer height, infill 20% GRID

What is right, 20% or 40%, Grid or Recilinear?
What is used in the factory?

prusa3d-bb, 2018-08-01:
Hello. 

Please use 20% and Grid infill. Meanwhile I will let my colleague know to correct the descriptions - thanks for letting us know about it. 

S pozdravem / Best Regards / Mit freundlichen Grüßen

—
Peter Mencák
Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

188/7a Partyzánská, 17000, Prague

shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

--- end of original message ---

3d-gussner, 2018-08-01:
Thanks for the fast reply. I think the gcode and amf file need to be updated.

ManuGithubSteam, 2018-09-06:
20 % is fine. Its much lighter that way and it is sturdy enough.

closed 2022-05-10 by Prusa-Support