prusa3d/Original-Prusa-i3 #91: Mk3 new 45 degrees part fan noise

max05210238, 2018-08-04:
Just printed and installed the new 45 degrees part fan assembly, and I think the part fan doesn't like that very much, it will makes a none periodically noise, it sounds like the fan bearing dosen’t seat right, I’m sure this noise is not the noise cost by the PWM speed control that everyone is talking about, and yes, I got that PWM noise too, it will happen when I use the PET setting to print PETG, so does anybody experience this 45 degrees part fan noise?

prusa3d-bb, 2018-08-06:
Hello,

could you please confirm if this still occurs with the fan check off as the upgrade should not cause any noise in the fan.

Best regardsDouglas Whyte
Customer Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
188/7a Partyzanska, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

--- end of original message ---

max05210238, 2018-08-09:
Ok, I had checked that, and yes, the noise is coming from the fan, and I had a 26 Hr print the other day, 25 Hr in, the fan noise is so big that I can't hear the stepper motor, the printer stops, and show me part fan error, I exam the fan, it still turns freely and no debris in it, resume the print, nothing wrong this time, but I switched back to the 90 degrees fan setup for now.

There's a recommendation I'd like to make, maybe let the part fan facing up instead facing down will help?

prusa3d-bb, 2018-08-10:
Hi there,

you might want to check the zipties are not pinching the wire causing this as this is possible as i assure you we have done this update on our printers here in the office and we dont get this issue, please let me know if any different after loosening the zipties.

Best regardsDouglas Whyte
Customer Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
188/7a Partyzanska, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

prusa3d-bb, 2018-09-07:
Hello,

so are you still getting the noise on the fan then ?

Best regardsDouglas Whyte
Customer Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
188/7a Partyzanska, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

Absolutely. It is awkward to let the fan blades hang this way.

—

--- end of original message ---

max05210238, 2018-09-08:
Hi Douglas,

After that little accident, I printed several prints without any issues, maybe the fan had some bad mood and decided to give me a hard time that day, LOL.

Anyhow, I’m really not feeling comfortable with the fan hanging that way, so I changed back to the 90 degree setting with a quite cleverly designed fan duct, see link below

https://www.thingiverse.com/thing:3063554

Used this setting more then 100 Hours of print time with no issues

prusa3d-bb, 2018-09-11:
HI there,

ok i see if its better for you this way no problem but does seem a bit weird though.

Best regardsDouglas Whyte
Customer Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
188/7a Partyzanska, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

https://www.thingiverse.com/thing:3063554

bradcozine, 2018-09-22:
I'm also having the issue with PET filament printing and fan noise. The tone begins immediately upon printing and occurred from my very first PET print today. My printer is 10 days old and zipties were all installed loosely. Also, the fan made no noise during PLA printing.

I used the tune option during to troubleshoot and tried a range of fan speeds and the only effect was to change the pitch and volume of the tone. Then only time the tone went away was when I set the tune amount very low, around 25 and below. Turning the fan off stops the noise of course.

I'm running the 3.4.0 version of the MK3 firmware. Since today is my first time with PET on this printer, I don't know if the issue occurs with earlier firmware.

prusa3d-bb, 2018-09-26:
Hello,

if the fan is indeed a bit noisy could you please provide a video to see as i am not sure why it would do this only with PETG even though yes for this filament it would be a bit needed as opposed to PLA but still should not be loud.

Best regards / Cordialement / Mit freundlichen GrüßenDouglas Whyte
Customer Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
188/7a Partyzanska, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

grhawk, 2018-09-26:
Hi, I am having the same problem with the fan. I will send you a video in the evening. BTW, I wanted to share something that looks interesting. I removed the fun and I noticed that keeping the fan horizontal with the blades "facing the sky" the noise goes away. Moreover, it really looks a problem with the bearings since it looks like the rotor is out of the center and get back his position when the fan is rotated as described before.

max05210238, 2018-09-26:
Yes, I have the same issue as grhawk, and now even I have the set up for 90 degrees, my part cooling fan still making noise sometimes, sounds like a dry bearing grinding, but it will soon go away, and the printer gives me part fan error continuously, I need to turn off fan detection so I can print, I think my fan is shocked or it's defective from the get-go, need a new one for sure, and I already contact prusa customer service.

trevjonez, 2018-09-26:
same as? https://github.com/prusa3d/Prusa-Firmware/issues/460

max05210238, 2018-09-26:
@trevjonez Yes and no, I have that PWM noise while printing PETG, because the fan speed is low, but I now have a noise sounds like a real broked fan, don't know it that caused by the 45 degrees position? it sounds like a dry grinding noise, I think my fan bushing wears out already.

grhawk, 2018-09-26:
Hi, 
Link to the video with the fan noise. I was making the fan moving just using the reset button. Other than the noise a clear sign that the issue is because of something with the bearing is the increse in vibrations when the fan is turned with the hole toward the bottom. Hope can be helpful.

https://youtu.be/oGatW7K-g7U

prusa3d-bb, 2018-10-02:
Hello,

for us to process each of these issues could you please personally send us an email to info@prusa3d.com as i cant troubleshoot this like this for a group chat.

Best regards / Cordialement / Mit freundlichen GrüßenDouglas Whyte
Customer Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
188/7a Partyzanska, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

https://youtu.be/oGatW7K-g7U

webdevbrian, 2019-02-05:
Getting noise on the same fan, just as indicated above. I will send you folks an email as well.

prusa3d-bb, 2019-02-07:
Hello Brian,

ok thank you for the information and any pictures or videos if possible would help to see the issue.

Best regards / Cordialement / Mit freundlichen GrüßenDouglas Whyte
Customer Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
188/7a Partyzanska, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

xyipher, 2019-02-15:
I'm also getting considerable fan noise ever since the fan was set to 45 degrees. Its worst at the start of a print and quiets down a bit after running but still very noticeable. I think the fan is also making more noticeable noise when it does long fast moves across the X and/or Y axis. 

I think the problem is this particular fan does not like the fan blade pulling away from the bearing/bushing when at the 45 degrees causing the noise. I would bet if the entire fan assembly was flipped 180 degrees so the fan blade had some down pressure on the bearing/bushing it would not make the noise.

prusa3d-bb, 2019-02-19:
HI there,

could you please provide a video of these issues for me to see or hear the fan.

Best regards / Cordialement / Mit freundlichen GrüßenDouglas Whyte
Customer Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
188/7a Partyzanska, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

xyipher, 2019-02-19:
The noise starts around 6 seconds, I caught it at the end of the first
layer. First layer it is quiet as the fan is not running during that time.
I tried to tap the fan while it was running so you could hear the noise
stop but it did't work as well as I was hoping. You can also notice the X/Y
movements that it really makes noise. Let me know if you need anything else.

Video: https://drive.google.com/open?id=19oKIQ78LeKydWIEp6bQhHR1x1EVKms0GTQ

Thanks

RiDDLeRN, 2019-02-19:
After I watched the attached video I can say that this is the exact behavior my printer has. No noise att all with the fan mounted vertical, but at 45 degrees it's horrible. I guess that the fan have a sleeve bearing and not a proper ball bearing.

max05210238, 2019-02-19:
Hi RiDDLeRn and xyipher,

Yes, I had the same issue around 6 months ago, sounds horrible at 45 degrees, even stop turning at low speed, I contact customer support, they shipped me a new fan, after this experience, I only mount my fan at 90 degrees, but after 6 months and 7 KG of filaments, it started acting up again, I think the fan is just not high quality, I took the new one apart, it's using brass bushing, no ball bearing, the grease dried up, so I put some "Super Lube 51010" in the brass bushing and assemble the fan back, works fine now, but don't know when the grease will dry up again.

I think the way to go is to buy a PWM controlled ball bearing fan, but I can't find one on Amazon, hope Prusa will do a recall, horrible fan.

AxonneQ, 2019-02-19:
I also experienced fan noise while printing PETG. When printing PLA no noise occured. This is because at 100% speed (for PLA) the noise is not present but as soon as the fan speed is below ~95%, the noise kicks in. (fan speed for PETG printing is set to around 50%). 

The newest firmware definitely helped because i didn't hear much noise when printing PETG today but still I also believe that the fan should be upgraded, It's way too noisy even when operating without issues.

prusa3d-bb, 2019-02-20:
Hi there,

i can send you a replacement fan for this but would need the order number as i cant seem to find the account details.

Best regards / Cordialement / Mit freundlichen GrüßenDouglas Whyte
Customer Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
188/7a Partyzanska, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

max05210238, 2019-02-20:
Hi prusa3d-bb,

Do you improve the fan? or I don't see any reason to get the same one, it will break again in 6 months for my kind of use....

prusa3d-bb, 2019-02-26:
Hello,

Sorry for the delay in getting back to you, we have been quite overwhelmed, following the release of our newsletter to over 100,000 people last week.

The newest version of firmware has a fix applied which adjusts the frequency of the print fan when at a lower speed setting. The noise should be fixed if you upgrade your firmware. If not please let us know your order number and, if possible, a video including the noise - we can provide a replacement.

Best regards,

—
Nikolai Zveznet
Data Scientist

_______________________________________

PRUSA Research

188/7a Partyzánská, 17000, Prague

+420 222 263 718

+421 220 570 305

Troubleshooting tutorials here!

Follow us on twitter and Facebook! 

Check out our blog!

________________________________________

Please rate our reply ...be generous :-)

--- original message ---

carpenda, 2019-04-22:
I just completed an upgrade to MK2.5S/MMU2S from MK2S/MMU and I just wanted you to know that I am hearing the same part fan noise that others have described.  I installed the latest firmware so that is not the issue.  Since my fan is two years old, it is not under warranty.  My question is, do you intend to offer an upgraded part fan in the future?  I am going to keep using this fan until it dies and then purchase a new one.

prusa3d-bb, 2019-04-24:
For the MK2.5-3s printers, the print fan will almost certainly not change. Unless, there was some trouble with getting the same fan supplied to us, which is very unlikely. 

Best regards,

—
Nikolai Zveznet
Data Scientist

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

--- original message ---

nahuelcarou, 2019-08-02:
Hello, im also getting noise at the 45° Fan. My print is brand new, has like 30 Hours printing and just startet with the noise. Could be the fan? Thanks!

nahuelcarou, 2019-08-02:
Hello All, I have solved the problem in my case: The problem is OIL. I ve just detached the Prusa Sticker on the fan and added some light oil. Now works perfect.  It should not happens.... Its a Prusa. One  point less there.

xyipher, 2019-08-02:
Although oil may solve the problem, a fan with a sleeve bushing should never need oil over its lifetime. If it did need oil there would most likely be a hole and a sticker saying so. 

The problem is one or both these things:
1. Cheaply made part
2. Was not meant to be mounted in the direction it is. Some fans in their tech specs have actual directions on how they can and cannot be mounted to perform as intended.

zveznet, 2019-08-06:
> 
> 
> Hello All, I have solved the problem in my case: The problem is OIL. I ve just detached the Prusa Sticker on the fan and added some light oil. Now works perfect. It should not happens.... Its a Prusa. One point less there.

For issues like this one which is not firmware related and where the fan may be defective, it would be better to contact customer support directly via chat: https://shop.prusa3d.com.

PaulGameDev, 2019-08-07:
The Prusa 5015s use sleeve bearings, which are indeed not meant to be used at 45°, but actually only in a vertical orientation, like they used to in the original MK3 design. I noticed this to be a problem after my fan died after ~1 month of usage. I received a replacement fan from Prusa yesterday, and I think in this configuration it won't take too long until it starts to get crunchy again. You might say you can fix this by oiling the fan from time to time, but this is not what the fan is designed for, and in my opinion more hacky than a proper fix, which is why I would recommend to switch to ball bearing fans for the 45° orientation

atary75, 2019-11-20:
I just did the MK3 to MK3S upgrade and now I've a noisy part fan!

Any solution to this issue?

jpf11, 2020-03-04:
I just assembled my mk3s and the fan is noisy.  Sounds like something clicking or hitting ever so slightly.  It kind of varies as it prints so I'm guessing something is out of spec and loose, allowing some part to vibrate.  I'm printing PETG at the moment with fan speed range of 20-50% in case that's relevant.  I haven't printed any other materials yet.

Pace17881, 2020-05-01:
Same problem here. I replaced the fan with a new from Prusa. This one shows the same problems over time :(

drelich, 2020-10-08:
Hilarious. Several years later and the problem still hasn't been addressed. 

Installing a fan in a way that the rotating part basically slides/hangs off its bearing is just stupid. Even with the fan off, any x-axis movement makes the rotating part rattle, creating unpleasant noise. I get it, on Prusa's print farm, nobody cares because the overall noise there is overwhelming. But we're talking consumer printer, which many hobbyist use in their bedrooms.

The whole printer is a piece of vibrating/noisy machine, especially if you want to print slightly faster. Got the printer for only a few days and I am already starting to question my decision to throw that kind of money at something that's been around for a few years and still suffers from elementary mechanical problems...

vsigler, 2020-11-02:
Just came here as the fan got terribly noisy. I've had MK2S for about 3 years, only about half a year ago upgraded to mk2.5 with the rotated fan, which now got terribly noisy. Not just the fan itself, but (even more) also when moving the X axis. I am looking to replace it. I've already dismounted it, mounted again, tightened all screws, but it is really the fan rotating part that's causing the noise. What were the reasons for the rotated position? I'm thinking of creating an upright mount, or even replacing the fan with something quiet. Just not sure if there isn't anything lose here in doing so.

CaptainFalcon92, 2021-03-11:
Hello people.

Letting you know that my 45° mounted blower fan is also making crunchy noises. 
To me that's the sound of worn out bearings. I Also believe the fan as not been designed to sit at an angle and is failing prematurely. It sounds like a barrel of gravel.

My right standing mk2 fan is still spinning loudly yet consistently.

Maybe flipping the fan on it's back, blades in, solid face out would extends bearing life ?

goyetus, 2022-05-02:
Any update in this problem?   Im with the same issue in two original prusa layer fans........

No solution?    Regards!

Prusa-Support, 2022-05-19:
Hello! Sorry for intervening on this thread so late. From what I read on this thread, it could have been still related with this thread: https://github.com/prusa3d/Prusa-Firmware/issues/1516. The fans generate at low RPM so much noise on the signal line that it is hard to distinguish if it is spinning or not, to be able to get some "readable" values the fan is forced to spin up. 

However, if the noise is more "continuous", it might be related with the inner mechanics of the fan (for example, inner bearing). Or even with the assembly of the fan itself (for example, this nut might be missing, or perhaps not pulled in with the screw-pulling technique: https://cdn.help.prusa3d.com/wp-content/uploads/840ffb9d3cab8282_painted.jpeg, hence causing a loose part that would vibrate during printing movements or just with the fan spinning. 

In case any of you is still experiencing issues, can you please contact me via email info@prusa3d.com mentioning me ("attn: Alessandro), with some videos of the noise? From here I can better assess the cases individually, as several causes could have been behind it - if it is a regular noise or anything related with the mechanics/assembly. I'll close the thread for now, but please do let me know if you are experiencing still issues via email. 

Alessandro Pantaleo
Prusa Research

CaptainFalcon92, 2022-05-19:
@Prusa-Support Hello.
This is **definitively** the fan bearing. The bearing **dies prematurely** because it sits **at 45°**.
**These fans are not design for being angled this way**. Its either straight up or laying down.
It's my **third fan i replace on my MK3** while my straight up fan on the MK2 is still the original.
I design an upright fan should, I think it should have been addressed by now.

goyetus, 2022-05-19:
Im with you @CaptainFalcon92 
I Have three fans,  all with the same problem.   Only 1 of them RMA.  
I send this info too at info@prusa3d.com   

I created a video here :  https://www.youtube.com/watch?v=dR1rqG_DDTo

It seems a Sleeve Bearing sound problem........   I would love to use ball bearing 5015 official fans..... insted of Sleeve bearing.

goyetus, 2022-05-19:
https://user-images.githubusercontent.com/24479700/169417621-a11b2edc-f26a-4543-bb71-0ed6db697a74.mp4

vsigler, 2022-05-20:
I agree with the above that the cause of this is excessive wear of the bearings due to the 45° angle the fan was not designed for.
I only got this issue once I upgraded to MK2S that has this fan shroud instead of the original one with vertical fan mount.

I am not having this issue at the moment only because I switched to a new MK3S, but expect it to develop the problem soon.

Prusa-Support, 2022-05-22:
Hello! Thank you for the additional feedback, in case of any specific cases to review (warranty, etc.), please let me know via email info@prusa3d.com mentioning me. But let me know of course also about the other aspects I mentioned in the earlier comment. 

I'll research with devs on the possible correlation between 45 degrees orientation and increasing wear. I haven't found info on this so far, but I'll certainly review this with our devs and collect more info. I'll keep the issue open so that anyone can add additional comments and feedback in the meantime. 

Alessandro Pantaleo
Prusa Research

goyetus, 2022-06-23:
> @goyetus thanks for sharing this video. The one i shelved is so much worse. Letme record it too https://www.youtube.com/watch?v=Y89l_YcBooo

UFFFFF your is so much worse as you said !!!!
@Prusa-Support   check this video too.   Same problem as  mine.    Fan bearings going crazy.

Mine is also doing sound without fan rotation ,   only with the printer axys moves. 

thanks for sharing!!!

Prusa-Support, 2022-06-24:
Hello :) 
@CaptainFalcon92 your video looked more related with X-axis movement specifically, the fan didn't do the noise during Y-axis movement. Can you please double-check when the print fan is off and X-axis is moving, for example the 1st layer calibration print? 

@goyetus so you have it when the axis moves, can you please check if the fan itself might have loosened over time? For example losing this nut, or just having the structure loose: https://cdn.help.prusa3d.com/wp-content/uploads/840ffb9d3cab8282_painted.jpeg can explain the noises occurring during movement, I remember experiencing it in person when I was new to 3D printing and getting scared by the sudden noise. :) 

Let me know, we'll review it in more detail but these two cases related with movement seemed to be more assembly-related. 

Alessandro Pantaleo
Prusa Research

goyetus, 2022-06-24:
> Hello :) @CaptainFalcon92 your video looked more related with X-axis movement specifically, the fan didn't do the noise during Y-axis movement. Can you please double-check when the print fan is off [...]

Mine was not loose at all.  It was a new fan in a new build.   I checked about 2-3 times that all was correct.  
Im about 5 years 3d-printing.  Print noise is inside my head as a 56k modem.  

It seems that noise comes from the fan in 45º position.  It stops or gets better in 90º or 0º .

CaptainFalcon92, 2022-06-25:
Hello @Prusa-Support 

I think a couple years of crunchy noisy prints is more than a double check. 😁
The printer is used for company prototyping and monthly serviced. This fan bearing is wrecked, ain't nothing lose for sure. 

It is so much an actual problem that we did "upgraded" our mk2s fan shroud from 90° up-straight to leaning 45° last year (mk2.3 project), and it also had ground the mk2 fan. 
I managed to prevent total death by having it back up again. Only time will tell how much it has shorten its lifespan.

Meanwhile my mk3s has killed its second fan bearing already (from prusa's shop). I might order cheaper spares one elsewhere as they keep on dying anyway.

Fitting a modified mk2 fan shroud on the mk3 is probably possible. I'll probably chop a new fan shroud for it some day.

The reality of it is that we came to accept the fact that this fan is now a short term consumable.
The loom has been fitted with a rapid Molex 51005/51006 connector so i'll simply swap for a 4th fan quicker next time. 

@goyetus That's how i was able to put back in our previous shredded fan for demonstration that quickly.

goyetus, 2022-07-19:
> Hello @CaptainFalcon92, thank you for the feedback! We'll further review with devs, keeping all feedback in mind - other possible causes we found on past cases: possible debris, hair, or possible over-tightening of the bolts thus slightly breaking the plastic covering the fan. We'll of course keep in mind all feedback and review in more detail our past print fan failures with 45 degrees compared to vertical (early MK3, MINI).
> 
> Alessandro Pantaleo Prusa Research

Man..... I have the problem in THREE new fans installed in less of two months.....
It´s not hair or dust.

- A Sleeve Bearing fan is made to work at Horizontal Plane ,  not 45º.
- A Ball Bearing fan ,  can work in any position.  
- The noise generated,  unless the fan is not operating,  is cause the sleeve bearing "cracking" inside the house at 45º.  

https://blog.fandis.com/en/cooling/difference-between-ball-bearings-and-sleeve-bearings-compact-fans/

agravelot, 2022-10-08:
I have exactly the same sound as @goyetus, whatever the orientation or the fan, except at 100% speed.

Got one in RMA, it's worse with the new one.

https://user-images.githubusercontent.com/13699253/194730840-887d4eb4-9fe7-4068-a86e-22f2cb8d26b8.MOV
https://user-images.githubusercontent.com/13699253/194730842-900e3341-f13a-4d39-b0c2-e33e52f70c35.MOV
https://user-images.githubusercontent.com/13699253/194730853-33709f20-0622-467a-a86d-d32ef33b040e.MOV

Prusa-Support, 2022-10-21:
Hello, thank you for the additional comments, all noted for our devs to further review. It might be also related with the PWM thing, @goyetus if you know of anyone more with the issue feel free to direct them to our technical support, we can look into it case by case, and collect additional feedback, always an useful thing. I'll update here in case of more info from our devs. 

Alessandro Pantaleo
Prusa Research