prusa3d/Original-Prusa-i3 #93: Model of the MK2(S) y-carraige

KasperADK, 2018-08-10:
Im looking for a model (stl, dxf or likely) of the Y-carraige used in MK2(S). Need the files for reference in Fusion360. Can't find them anywhere, neither could you support chat, who directed me here. 

Thanks :-)

prusa3d-bb, 2018-08-10:
Hello,

Here is the link you need. Downloads contain both STLs and SCAD files.
https://www.prusa3d.com/prusa-i3-printable-parts/

Best regards,
William Holm
Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

188/7a Partyzánská, 17000, Prague

shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

--- original message ---

--- end of original message ---

DeJennes, 2018-08-13:
@KasperADK there is a DXF file for the frame. The Y carriage is included there

lecktor, 2018-08-19:
I've added separate file (+ some modifications) here: https://github.com/lecktor/lecktor3D/tree/master/heated_bed
green -> engrave (0.5mm)
red -> cut first
white -> cut second