prusa3d/Original-Prusa-i3 #95: Need PRINT-SETTINGS.md in Printed-Parts/

ayberkozgur, 2018-08-14:
Pretty easy. Just an `.md` with required print settings to avoid confusion.

ManuGithubSteam, 2018-09-06:
Its on the website and in the Build manual on the website.
Its in every build guide on the website and the FAQ if im not mistaken.
And its in the printed version of the manual

I dont see a need for more places. :-)

ayberkozgur, 2018-09-06:
None of these sources are properly version controlled, i.e you don't know which settings correspond to which commit that contains the STLs.

I would go as far as providing gcodes and even config bundles right here in the repo, all version controlled of course.

Every new "mirror" of the same source/document is extra work and, if not properly maintained, a recipe for disaster. I don't know why links at https://www.prusa3d.com/prusa-i3-printable-parts/ don't point to releases here.

ManuGithubSteam, 2018-09-06:
Hmm i see your point. Although i think stl is not so critical that every minior detail change will need to be reflected in print settings like infill pattern and temperature. My experience goes more in the direction of stick with one filament, Tune the hell out of it and then print with that filament with the best settings regardless if i print a 1cm piece or a 20x15 cm cube. So i think it is not necessary to have it that tight under control. However links to the releases here would be nice.