prusa3d/Original-Prusa-i3 #96: LCD cover reset button snaps off

probonopd, 2018-08-19:
The reset button part in the LCD cover of my original MK2S snapped off after a year or so. 

I think it is because the cylindrical part going from the tactile switch to the cover is __not long enough__ and the material cannot flex that much and will eventually break.

Making the cylindrical part that goes down to the button longer (so that it just barely touches the button) would probably solve this.

[image]

probonopd, 2018-10-20:
Here is a replacement button to insert into the existing housing without having to reprint the whole cover.

[image]

[Download button.zip](https://github.com/prusa3d/Original-Prusa-i3/files/2498252/button.zip)

Serveurperso, 2019-06-15:
LOL 3D printing and repair is fun thanks!