prusa3d/Original-Prusa-i3 #98: Filament Runout sensor?

npgoalkeeper, 2018-09-06:
Instead of using a PINDA probe or an optical laser to detect filament, why not use a wheel and 2 bearings with a magnetic encoder such as an AS5601P to detect the filament? It can detect speed and the presence of the filament easily, as long as its moving. This shouldn't matter with any of the printer's functions.

Moorviper, 2018-09-06:
Then you have to set the  wheel for each filament. 
the laser is a good way.
I am currently building my own multimaterial device (uses 2 laser mouse sensor and detect in addition the filament by color by another sensor.
It also checks under the nozzle if the color is clean and then wipe the nozzle  with a brush.
Then i need no purge tower or wipe into infill on my printers.

ManuGithubSteam, 2018-09-06:
I sencond the laser :-)

npgoalkeeper, 2018-09-07:
Why does it need to be set for each filament? It can be like other printers, using a spring-loaded idler gear. As long as the wheel spins and matches the desired extrusion rate, it should be okay. I don't see how the as5047p (not the as5601 lol) would react any different to a flexible than PLA.

npgoalkeeper, 2018-09-07:
The only thing I do see, however, is the cost. It is heavier and costs 10x as much for this setup rather than a laser. Also, lasers throw out false positives.

Moorviper, 2018-09-07:
I got two prusa laser sensor clones for around 5€ inkl shipping i don't think you are getting much cheaper.
The cheapest magnetic encoder on digikey is around 4,5€ alone

bdynamic, 2018-09-07:
@Moorviper Could you maybe share your source? Ali has them for about 11$...

jltx1, 2018-09-09:
> Yes now there was a sale week last week.
> actually it cost 8,47€/$9,80 ok -80% within the app was a good deal

My fix:
https://www.thingiverse.com/thing:3091625 <https://www.thingiverse.com/thing:3091625>

npgoalkeeper, 2018-09-14:
That defeats the point of why I opened this issue, however. The PINDA probe cant detect jams.

closed 2018-11-23 by npgoalkeeper