prusa3d/Original-Prusa-i3 #99: Disable filament sensor mid-print

DesignFlaw06, 2018-09-06:
I ran into an issue with a print last night where the filament sensor incorrectly thought filament had run out.     It was black PETG and I understand that can be a problem for the filament sensor.  The problem is that I couldn't disable the filament sensor mid-print so I had to scrap what already been printed. 

It would be nice if there's an option to disable the filament sensor during a print in the event that it is a problematic filament.

ManuGithubSteam, 2018-09-06:
Hi

Funny, i just had the same issue last night too. I used Black PETG Filament from Real Filament.
However i was able to "reload" the filament and continue printing. 

Moreover, it is possible to shut off the filament sensor mid print. I did just that.
Its in the menu under Tuning i believe.

What firmware version do u use ?

Maybe its not there in your version...

DesignFlaw06, 2018-09-06:
The beeping woke me up at about 5 AM.  I have to admit I wasn't entirely conscious when I went to look at it.  I could have continued the print, but I know that filament has that issue so it would just happen again.  I simply forgot to do it when I started.  I'll look in the Tune menu when I get home  during the print.  

Sorry for the false issue.

closed 2018-09-06 by DesignFlaw06