prusa3d/Original-Prusa-i3 #101: Newest STL files unable to download.

beneq69, 2018-09-28:
Newest STL files for Prusa MK3 are unable to download and print Only Einsy-base is available

stahlfabrik, 2018-09-28:
I can download the zip file - and all new stl files are in it. The openscad files are not updated though, but that is another issue

beneq69, 2018-09-28:
Original-Prusa-i3/Printed-Parts/stl/ files added yesterday.

stahlfabrik, 2018-09-28:
Yes. What is the issue?

beneq69, 2018-09-28:
So ill ask different way. Which program i have to use to open that new stl files? I tried Slicer and Simplify3D and after downloaded files i can only files: "Z-axis-bottom", "Einsy-base", "x-end-motor", "x-carriage". All others new files I can't open.

stahlfabrik, 2018-09-28:
That is strange, I tried to open X-end-idler in Slic3r. Works like a charm. Also the Y motor holder works just fine. I guess that you have a corrupted file?!

stahlfabrik, 2018-09-28:
So have you downloaded the zip file?

beneq69, 2018-09-28:
Yes, Download one more time couple minutes ego and I can open files right now. Thank You for help

closed 2018-09-28 by beneq69