prusa3d/Original-Prusa-i3 #105: OpenSCAD, incorrect naming of scad versus stl file

mariusvw, 2018-10-11:
The idea is to automate the rendering with a shell script, the STL files are not named correctly compared to the SCAD files.

STL: Heatbed-cable-clip_for_8mm_sleeve.stl

If someone can tell me which SCAD file is the correct one to render this target I could fix this.

bvarner, 2019-04-12:
:+1 The inconsistencies are driving me crazy as well.

mariusvw, 2019-04-13:
@josefprusa not sure if you read this Josef, but it would be nice to work together on Github with your team. So far I came up with several improvements but get stuck at being able to merge it.

If I look at the PR-count it looks that things aren't sorting out. If any help from my side is wanted, don't hesitate to contact me.