prusa3d/Original-Prusa-i3 #112: MK3, extruder fan part, append hook on the right side

mariusvw, 2018-10-27:
Sometimes it happens that the cooling part does not fit straight, adding a little hook on the richt site to clip it on will help supporting it to be straight.

(already working on this, will soon make PR)

mariusvw, 2018-10-29:
Hello Marina,

It is not a problem caused with the ordered parts, it occurs when printing your own parts.
Since the right side (left in the picture) has no attaching to the extruder body it "hangs" a bit more.
When you look closely you see that it is slightly not at te same level compared to both sides.

[image]

Adding a little "hook" to the fan part will force it to stay in place, you still have one bolt on the left side to lock it in place.

You can see my idea more as an upgrade than as a problem.

Almost finished the change so I will be adding a pull request referring to this issue soon.

-Marius

prusa3d-bb, 2018-10-30:
Hello,

Thank you for your email. Unfortunately I could not open the picture.

Best regards,

—
Marina Curzi
Customer Support 

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

--- end of original message ---

rrauenza, 2018-10-31:
I believe what @mariusvw is saying is that because the parts cooler nozzle is only screwed in with one screw, and that screw is only on one side, the parts cooler tends to hang a little at an angle.  I've also have seen this.

mariusvw, 2018-11-01:
@prusa3d-bb what @rrauenza describes is exactly what I am trying to say.

I almost finished an upgrade for this on the SCAD files, will make a PR.