prusa3d/Original-Prusa-i3 #116: Einsy-base unsupported overhang on cable entrance

aemileski, 2018-11-25:
There is a 45 degree ramp under the cable entrance, but it does not extend far enough.

It appears as though the cable entrance was extended, but the support ramp wasn't adjusted.

[image]

prusa3d-bb, 2018-11-26:
Hello,

thank you for your feedback. I will give your feedback to my colegue from hardware DEV. 

Kind regards,

—
Šimon Doubek
Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

188/7a Partyzánská, 17000, Prague

shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

--- original message ---

--- end of original message ---

tkircher, 2019-06-11:
I had to fix this in the process of upgrading to the MK2.5S. A simple solution is to add extruded triangular supports under each part that's unsupported. Another issue is that the resolution on Prusa's models is very low. The number of facets they specify for cylinders is too low by a factor of at least four. It should not be possible to hear the printer changing direction in discrete steps. Fortunately the source code is available.

N3X15, 2019-06-11:
Careful, a bunch of parts are desynced from the code, last time I checked.

tkircher, 2019-06-11:
If you need to check whether parts are synchronized, render the SCAD file then subtract the two meshes. Ordinarily I'd also suggest modifying the released STL files directly, but it turns out that Prusa's shipped parts and the ones they provide for printing aren't the same.

prusa3d-bb, 2019-06-13:
Hello Thomas,

the 3D printed parts are same as we are printing there in Prague as the parts available on the GitHub or our webpage.
I will again pass your feedback to our development.

Kind regards,

—
Šimon Doubek
Customer Support
_______________________________________
PRUSA Research
188/7a Partyzánská, 17000, Prague
+420 222 263 718
+421 220 570 305

Troubleshooting tutorials here!
Follow us on twitter and Facebook! 
Check out our blog!
________________________________________
Please rate our reply ...be generous :-)

--- original message ---