prusa3d/Original-Prusa-i3 #117: Is this only for MK2 or is the description wrong?

Spacemarine2018, 2019-01-15:
The discription says "Original Prusa i3 MK2", yet most of the content here deals with the MK3.

prusa3d-bb, 2019-01-15:
Hello,

Can you send me a link please? I'm not sure what are you saying.

Best regards.

—

Alberto López Customer Support
Spanish, English, Portuguese

_______________________________________

PRUSA Research

188/7a Partyzánská, 17000, Prague

+420 222 263 718

+421 220 570 305

Troubleshooting tutorials here!

Follow us on twitter and Facebook! 

Check out our blog!

________________________________________
Please rate our reply ...be generous :-)

--- original message ---
On January 15, 2019, 12:47 PM GMT+1 Spacemarine2018 wrote:

The discription says "Original Prusa i3 MK2", yet most of the content here deals with the MK3.

—

--- end of original message ---

Moorviper, 2019-01-15:
the mk2 is the default branch

prusa3d-bb, 2019-01-16:
Hello,

We appreciate the help and feedback from the community, you can report a problem in github and our development team will take a look.

Best regards.

—

Alberto López Customer Support
Spanish, English, Portuguese

_______________________________________

PRUSA Research

188/7a Partyzánská, 17000, Prague

+420 222 263 718

+421 220 570 305

Troubleshooting tutorials here!

Follow us on twitter and Facebook! 

Check out our blog!

________________________________________
Please rate our reply ...be generous :-)

--- original message ---
On January 15, 2019, 9:58 PM GMT+1 Daniel Krah wrote:

the mk2 is the default branch

—

--- end of original message ---

Moorviper, 2019-01-16:
there is no problem he has just to read:
https://guides.github.com/activities/hello-world/
and understand what github is and how it works.

closed 2020-08-09 by Spacemarine2018